zigpy/zigpy #1351: OTA from z2m_remote_index not filtering manufacturerer name correctly resulting in bricked devices

skgsergio, 2024-03-03:
# Summary
I'm using `z2m_remote_index` and my two Lidl Silvercrest Smart Plug, model `TS011F`
manufacturerer `_TZ3000_kdi2o9m6` (https://zigbee.blakadder.com/Lidl_HG06337.html), wanted to update from `0x00000042` to `0x00000057`.

I went ahead and updated one of them and it bricked.

# Root cause?
Checking the index the only firmware with version `0x00000057` in the index is filtered by `manufacturerName` (a list of manufacturer names) and seems to correspond to Thermostatic Valves (https://zigbee.blakadder.com/Moes_ZTRV-ZX-TV02-EN.html, https://zigbee.blakadder.com/Hysen_HY369.html). The manufacturer of my device is not listed there.

Looks like there is a bug that is making `zigpy` to ignore the `manufacturerName` filter and is that it is actually trying to check for `manufacturerNames` (note the last extra `s`): https://github.com/zigpy/zigpy/blob/0.63.3/zigpy/ota/providers.py#L470

Z2M through `zigbee-herdsman-converters` library uses this property without the `s`: https://github.com/Koenkk/zigbee-herdsman-converters/blob/v18.42.0/src/lib/ota/zigbeeOTA.ts#L65

# Details
zigpy config:
```yaml
    z2m_remote_index: "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json"
```

Firmware entry in the index:
```
    {
        "fileVersion": 87,
        "fileSize": 234174,
        "manufacturerCode": 4098,
        "manufacturerName": [
            "_TZE200_kds0pmmv",
            "_TZE200_ckud7u2l",
            "_TZE200_cwnjrr72",
            "_TZE200_ywdxldoj",
            "_TZE200_cpmgn2cf"
        ],
        "imageType": 5634,
        "sha512": "1113486a67403d922fc4a462b7477e0a0986fa8862a89b74ab8c5eac7b4ecb31c30f1da9ae1bbc1d6d50ae1fed66d38c37a1834ab71fd384c0555002783aba13",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Tuya/1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin",
        "path": "images/Tuya/1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin"
    },
```

skgsergio, 2024-03-03:
I don't like mention someone directly, but I think is important to fix this bug ASAP to prevent more bricked devices so... pinging @puddly and @dmulcahey as the author and approver of the #1340 PR that introduced the bug.

Hedda, 2024-03-03:
FYI, believe the risk of bricking Tuya devices was mentioned in https://github.com/zigpy/zigpy/pull/1340

Hedda, 2024-03-03:
> FYI, believe the risk of bricking Tuya devices was mentioned in #1340

And also in https://github.com/home-assistant/core/pull/111159

skgsergio, 2024-03-03:
@Hedda in fact the risk mentioned there is when you use the ota directory without metadata, which precisely z2m index attempts to mitigate. And I've used the z2m index.

Yes, I know there is no 0 risk due to the shitty Tuya ecosystem in particular and how vendors handle updates and access to them.

I'm not blaming anyone here, I was willing to test, knew the risk and that's why I've updated 1 device at a time and tested the result.

But in any case there is bug in the logic of the z2m index metadata loading and I'm reporting it. Your comment doesn't provide any value, if there is a bug can't I report it?

closed 2024-03-03 by dmulcahey