zigpy/zigpy #1357: Potential copyright issues with inclusion of OTA update files

gspr, 2024-03-04:
I see that PR #1340 ended up in the inclusion of OTA update files from some manufacturers into the ZigPy git repository.

As things currently stand, ZigPy claims to be under the GPL-3+ license. That license surely does not apply to these update files. The project should clearly document which files the license applies to, and which files are under a different license. Or, ideally, the OTA update files should be excised from the repository.

This is important both to protect the ZigPy project and users using ZigPy under the terms of the GPL.

MattWestb, 2024-03-04:
The file` tests/ota/files/ikea/mgm210l-light-cws-cv-rgbw_release_prod_v268572245_3ae78af7-14fd-44df-bca2-6d366f2e9d02.ota` is not downloaded from IKEAs feed its sniffed with wireshark then Dirigera hub have updating one device and rebuild as one plane Silabs OTA file that is signed by IKEA.
From the Principe i agree with you nd best is letting Z2M having the dirty OTA files not not in this organisations GIT for not getting problems in the future.
But we have spreading very much dumped firmware and bootloaders from IKEA devices so users can recovering there device then IKEA have making bad firmware's.

gspr, 2024-03-04:
> The file` tests/ota/files/ikea/mgm210l-light-cws-cv-rgbw_release_prod_v268572245_3ae78af7-14fd-44df-bca2-6d366f2e9d02.ota` is not downloaded from IKEAs feed its sniffed with wireshark then Dirigera [...]

How the file was obtained (download, sniffing, or otherwise) is irrelevant here.

pipiche38, 2024-03-04:
Even in you hack via a Man in the Middle, the firmware is still the property of the manufactuer, doesn't really matter how you get it.
Without the rights from the manufacturer I don't think you are allow to put into a any License mecanism

puddly, 2024-03-04:
Are unit tests considered code covered by the GPL? The final PyPI package does not include any test files.

gspr, 2024-03-04:
> Are unit tests considered code covered by the GPL? The final PyPI package does not include any test files.

IANAL, but I don't think the GPL (or any other popular license) makes the distinction.

The way I see the matter, there's two separate problems:

1. The copyright holder for the OTA update files do not allow you to redistribute them. This is a problem between the ZigPy authors and those upstream copyright holders. (Incidentally, this probably also puts in you violation of GitHub's terms of service).
2. Downstream users and/or redistributors of ZigPy approach the ZigPy git repository and think "cool – I wanna use this, let's look at which license the authors have put the code under". They then see only the single notice of ZigPy being GPL3+. With no other notices or specifications anywhere in the repository, they are led to believe (reasonably, in my mind) that *every* file in the repository is GPL3+. That is clearly not true. This is a problem between ZigPy and its users.

Either way, a serious project needs to take issues like these seriously. Ideally, the license for each file in the repository should be clearly stated. But the easiest fix would be to simply drop the files for which you are not absolutely certain of the license of (which I assume includes the OTA update files).

PR #1358 (2024-03-04): Download blobs during testing instead of keeping them in-tree
Closes #1357, since the OTA images are publicly available.

They'll be kept around in-tree but not in the Git repo.
closed 2024-03-04 by puddly