zigpy/zigpy #1366: Aqara E1 TRV Unsupported time cluster

Mikaka27, 2024-03-11:
Hi,

In recent home assistant update 2024.03.0, I've added this to configuration.yaml.
```
zha:
  zigpy_config:
    ota:
      z2m_remote_index: https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json
```

I have Aqara E1 TRV (lumi.airrtc.agl001), which I updated to newest firmware.
Since then I have problems with schedules on the TRV that start and stop at incorrect time.

I've noticed both during device re/pairing and manually triggering time update on the device (via zha_toolkit) that I get this:
```
2024-03-11 17:08:34.927 INFO (MainThread) [custom_components.zha_toolkit] Running ZHA Toolkit service: <ServiceCall zha_toolkit.misc_settime (c:01HRQ4BQVEWXNGTQ0TW7WEVMY2): fail_exception=True, ieee=button.grzejnik_duzy_pokoj_zidentyfikuj_2>
2024-03-11 17:08:34.927 DEBUG (MainThread) [custom_components.zha_toolkit] Got hass.data['zha']/gateway <homeassistant.components.zha.core.gateway.ZHAGateway object at 0x7ffa0af956d0>
2024-03-11 17:08:34.944 DEBUG (MainThread) [custom_components.zha_toolkit] module is <module 'custom_components.zha_toolkit' from '/config/custom_components/zha_toolkit/__init__.py'>
2024-03-11 17:08:34.948 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Parameters '{'fail_exception': True, 'ieee': 'button.grzejnik_duzy_pokoj_zidentyfikuj_2'}'
2024-03-11 17:08:34.948 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Final manf 'None'
2024-03-11 17:08:34.949 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Found registry_entity RegistryEntry(entity_id='button.grzejnik_duzy_pokoj_zidentyfikuj_2', unique_id='54:ef:44:10:00:70:6c:d4-1-3', platform='zha', previous_unique_id=None, aliases=set(), area_id=None, capabilities=None, config_entry_id='84f21b7b375c70350fd68cfd961e3b82', device_class=None, device_id='cf5498d0f88bb7d57fa023bf9977c493', disabled_by=None, entity_category=<EntityCategory.DIAGNOSTIC: 'diagnostic'>, hidden_by=None, icon=None, id='4ec651fa25f1c511e62e475b50672352', has_entity_name=True, labels=set(), name='Grzejnik duży pokój Zidentyfikuj', options={'conversation': {'should_expose': False}}, original_device_class='identify', original_icon=None, original_name='Zidentyfikuj', supported_features=0, translation_key=None, unit_of_measurement=None)
2024-03-11 17:08:34.949 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Found registry_device DeviceEntry(area_id='salon', config_entries={'84f21b7b375c70350fd68cfd961e3b82'}, configuration_url=None, connections={('zigbee', '54:ef:44:10:00:70:6c:d4')}, disabled_by=None, entry_type=None, hw_version=None, id='cf5498d0f88bb7d57fa023bf9977c493', identifiers={('zha', '54:ef:44:10:00:70:6c:d4')}, labels=set(), manufacturer='LUMI', model='lumi.airrtc.agl001', name_by_user='Grzejnik duży pokój', name='LUMI lumi.airrtc.agl001', serial_number=None, suggested_area=None, sw_version=None, via_device_id=None, is_new=False)
2024-03-11 17:08:34.950 DEBUG (MainThread) [custom_components.zha_toolkit] 'ieee' parameter: 'button.grzejnik_duzy_pokoj_zidentyfikuj_2' -> IEEE Addr: '54:ef:44:10:00:70:6c:d4'
2024-03-11 17:08:34.950 DEBUG (MainThread) [custom_components.zha_toolkit] Default handler for misc_settime
2024-03-11 17:08:34.950 DEBUG (MainThread) [custom_components.zha_toolkit] Handler: <function command_handler_default at 0x7ffa0f80bc40>
2024-03-11 17:08:34.950 DEBUG (MainThread) [custom_components.zha_toolkit] running default command: <ServiceCall zha_toolkit.misc_settime (c:01HRQ4BQVEWXNGTQ0TW7WEVMY2): fail_exception=True, ieee=button.grzejnik_duzy_pokoj_zidentyfikuj_2>
2024-03-11 17:08:34.952 DEBUG (MainThread) [custom_components.zha_toolkit.default] Trying to import custom_components.zha_toolkit.misc to call misc_settime
2024-03-11 17:08:34.954 DEBUG (MainThread) [custom_components.zha_toolkit.misc] Default time zone Europe/Warsaw
2024-03-11 17:08:34.954 DEBUG (MainThread) [custom_components.zha_toolkit.misc] Next dst changes 2024-03-31 01:00:00 .. 2024-10-27 01:00:00 EPOCH 2000 765162000.0 .. 783306000.0
2024-03-11 17:08:34.954 DEBUG (MainThread) [custom_components.zha_toolkit.misc] Local 2024-03-31 03:00:00+02:00 2024-10-27 02:00:00+01:00 in Europe/Warsaw 3600.0 0.0
2024-03-11 17:08:34.955 DEBUG (MainThread) [custom_components.zha_toolkit.misc] UTC OFFSET: 3600.0  DST OFFSET: 3600.0
2024-03-11 17:08:34.955 DEBUG (MainThread) [custom_components.zha_toolkit.utils] IEEE for get_device: 54:ef:44:10:00:70:6c:d4 54:ef:44:10:00:70:6c:d4
2024-03-11 17:08:34.955 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Endpoint 1 found for cluster '10'
2024-03-11 17:08:53.463 DEBUG (MainThread) [custom_components.zha_toolkit] event_data {'zha_toolkit_version': 'v1.1.8', 'zigpy_version': '0.63.4', 'zigpy_rf_version': '0.12.1', 'ieee_org': 'button.grzejnik_duzy_pokoj_zidentyfikuj_2', 'ieee': '54:ef:44:10:00:70:6c:d4', 'command': 'misc_settime', 'command_data': None, 'start_time': '2024-03-11T16:08:34.950228+00:00', 'errors': [], 'params': {'dir': 0, 'tries': 1, 'expect_reply': True, 'args': [], 'read_before_write': True, 'read_after_write': True}, 'read_before': ({}, {0: 0, 1: 0, 2: 0, 3: 0, 4: 0, 5: 0}), 'result_write': Default_Response(command_id=2, status=<Status.UNSUPPORTED_CLUSTER: 195>), 'read_after': ({}, {0: 0, 1: 0, 2: 0, 3: 0, 4: 0, 5: 0}), 'success': True}
```

Problematic thing is response: **Status.UNSUPPORTED_CLUSTER: 195**. I now wonder if this is device's fault?
Can time cluster be unsupported even though it is displayed in device's available clusters?

puddly, 2024-03-11:
Can you remove the device from your network and re-join it so that it is fully re-initialized?

Mikaka27, 2024-03-11:
Sure here is the log, I see this **Status.UNSUPPORTED_CLUSTER: 195** along the way, but there is also one stack trace.
[zha device join.txt](https://github.com/zigpy/zigpy/files/14561840/zha.device.join.txt)

closed 2024-09-14 by github-actions[bot]