zigpy/zigpy #1382: OTA does not distinguish between up-to-date devices and devices without updates

puddly, 2024-04-12:
Our OTA handling doesn't distinguish between an update not being available due to the device being already up-to-date and no image existing. This is because we filter out "incompatible" OTA updates during the selection process.

One possible solution is to return the latest firmware image as opposed to `None`. This has one major downside, as we are as lazy as possible when sending HTTP requests but also as safe as possible when validating metadata. We can't be sure that an OTA image is compatible with a device until we download the metadata index *and* the entire OTA file. This is wasteful: we would download a full OTA image for every compatible device just to indicate that it is up-to-date.

Another possibility is to mark OTA images with metadata as "complete" or "incomplete", allowing us to skip downloading the image unless necessary. We can also cache firmware images to disk (keyed by their metadata), allowing us to implement the above version scheme without extraneous HTTP traffic.

MattWestb, 2024-05-07:
One thinking is way not storing OTA metadata in Zigbee.DB in one table (or if its better one OTA.DB) and only updating it one or two times daily and then can the system only cross checking of the tables of the device and the OTA metadata. Perhaps more often or manual for local stored OTA files so not need restarting HA for getting them "hocked".

Perhaps storing the the manufacture and image ID then device is being added (i think its already in the DB) and only do checking and download of this so not flooding downloads and still up to dated information.

You is knowing the database structure and what is possible and god for the system but i like DB things then its much better then doing large works "online / real time" all the time and wasting  CPU power all the time.

Thanks for great work done and using your brain very well !!!

puddly, 2024-08-05:
An OTA image cache in the database would be very useful. We can explore this in a future PR.

PR #1397 (2024-08-28): Pass all compatible firmware metadata when receiving an OTA upgrade request
Fixes #1382.

This PR replaces `get_ota_image` with `get_ota_images`, which returns all compatible OTA image metadata. For upgrades, this metadata will also contain the full firmware image. For downgrades, it will contain just metadata.

This allows applications to infer the state of the device with much more granularity:

1. If there are upgrades, the first one is what should be upgraded to.
2. If there are no upgrades but there are downgrades, there are two possibilities:
   1. The most recent downgrade's current fir [... cut]
closed 2024-08-28 by puddly