zigpy/zigpy #1390: Error with latest zigpy -, failed to initialize due to unexpected error -  AttributeError: 'AnonymousLVList' object has no attribute 'is_end_device'

pipiche38, 2024-04-26:
With the latest zigpy version I'm getting this error. Reverting to previous one
```
2024-04-24 20:13:12,940 WARNING :Device <Device model=None manuf=None nwk=0x88E4 ieee=00:04:74:00:00:82:a5:4f is_initialized=False> failed to initialize due to unexpected error
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 205, in initialize
    await self._initialize()
  File "/usr/local/lib/python3.10/site-packages/zigpy/util.py", line 131, in retry
    return await func()
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 234, in _initialize
    status, _, endpoints = await self.zdo.Active_EP_req(self.nwk)
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 314, in request
    if expect_reply and (self.node_desc is None or self.node_desc.is_end_device):
AttributeError: 'AnonymousLVList' object has no attribute 'is_end_device'
2024-04-24 20:13:12,996 WARNING :Device <Device model=None manuf=None nwk=0x88E4 ieee=00:04:74:00:00:82:a5:4f is_initialized=False> failed to initialize due to unexpected error
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 205, in initialize
    await self._initialize()
  File "/usr/local/lib/python3.10/site-packages/zigpy/util.py", line 131, in retry
    return await func()
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 234, in _initialize
    status, _, endpoints = await self.zdo.Active_EP_req(self.nwk)
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 314, in request
    if expect_reply and (self.node_desc is None or self.node_desc.is_end_device):
AttributeError: 'AnonymousLVList' object has no attribute 'is_end_device'
2024-04-24 20:13:13,225 INFO    :b'Unknown Window Covering Command: 06'
2024-04-24 20:13:13,268 WARNING :Device <Device model=None manuf=None nwk=0x88E4 ieee=00:04:74:00:00:82:a5:4f is_initialized=False> failed to initialize due to unexpected error
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 205, in initialize
    await self._initialize()
  File "/usr/local/lib/python3.10/site-packages/zigpy/util.py", line 131, in retry
    return await func()
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 234, in _initialize
    status, _, endpoints = await self.zdo.Active_EP_req(self.nwk)
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 314, in request
    if expect_reply and (self.node_desc is None or self.node_desc.is_end_device):
AttributeError: 'AnonymousLVList' object has no attribute 'is_end_device'
```

TheJulianJES, 2024-04-26:
So, 0.63.5 is fine, but 0.64.0 is broken for you?

TheJulianJES, 2024-04-26:
I'm seeing no changes in that version that would affect this.

What device is this?
`Device model=None manuf=None nwk=0x88E4 ieee=00:04:74:00:00:82:a5:4f`

pipiche38, 2024-04-27:
problem occured, when pairing a new device.
The device here showing the error is a 'Dimmer switch wo neutral' from Legrand

pipiche38, 2024-04-27:
Did get the same error with an INNR SP120 plug. Unfortunatly as soon as I enable the full zigpy _ bellows logs I'm not getting any more

TheJulianJES, 2024-04-28:
Are you sure this is caused by the zigpy dependency bump and not by bellows [0.38.2](https://github.com/zigpy/bellows/releases/tag/0.38.2)?

pipiche38, 2024-04-28:
> Are you sure this is caused by the zigpy dependency bump and not by bellows [0.38.2](https://github.com/zigpy/bellows/releases/tag/0.38.2)?

Indeed I'm upgrading both zigpy and bellows

TheJulianJES, 2024-04-28:
So, just upgrade zigpy by itself, check again, and then only update bellows by itself to check where the problem originates from.

pipiche38, 2024-04-30:
not able to duplicate on zigpy 0.64.0.
So I'll just upgrade the zigpy layer and wait for a more stable bellows part

closed 2024-04-30 by pipiche38