zigpy/zigpy #1391: Issue when loading AdvancedFileProvider from config file

simonfr, 2024-04-27:
Hi,

I'm trying to init AdvancedFileProvider using this configuration :
```
zigpy_config:
  ota:
    enabled: true
    z2m_remote_index: https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json
    allow_advanced_ota_dir: I understand I can *destroy* my devices by enabling OTA
      updates from files. Some OTA updates can be mistakenly applied to the
      wrong device, breaking it. I am consciously using this at my own risk.
    advanced_ota_dir: ./zigbee_ota
```

And I think something is missing to convert the `advanced_ota_dir` to `pathlib.Path`.

Here is the error : 
```
2024-04-27 15:01:45.045 DEBUG (MainThread) [zigpy.ota] Failed to load provider <zigpy.ota.providers.AdvancedFileProvider object at 0x7f8e5af860>
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/ota/__init__.py", line 333, in get_ota_image
    index = await self._load_provider_index(provider)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/util.py", line 491, in replacement
    return await tasks[key]
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/ota/__init__.py", line 309, in _load_provider_index
    return await provider.load_index()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/ota/providers.py", line 161, in load_index
    return [meta async for meta in self._load_index(session)]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/ota/providers.py", line 713, in _load_index
    for path in self.image_dir.rglob("*"):
                ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'rglob'
```

TheJulianJES, 2024-04-27:
Which zigpy version are you using? Did you try HA beta 2024.5.0b1 or later?

simonfr, 2024-04-29:
I'm using zigpy 0.63.5 (from HA 2024.4.4). Upgrading to latest beta works ! Can you show me why ? Thanks

puddly, 2024-04-29:
https://github.com/zigpy/zigpy/pull/1371 fixed it.

closed 2024-04-29 by puddly