zigpy/zigpy #1393: Listeners error: `RuntimeError: dictionary changed size during iteration`

puddly, 2024-04-29:
It seems like the `_listeners` dictionary can change size during iteration.

```python
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy_znp/api.py", line 1059, in request
    response = await response_future
               ^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError
​
During handling of the above exception, another exception occurred:
​
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 673, in _watchdog_loop
    await self.watchdog_feed()
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 655, in watchdog_feed
    await self._watchdog_feed()
  File "/usr/local/lib/python3.12/site-packages/zigpy_znp/zigbee/application.py", line 628, in _watchdog_feed
    await self._znp.request(c.SYS.Ping.Req())
  File "/usr/local/lib/python3.12/site-packages/zigpy_znp/api.py", line 1055, in request
    async with async_timeout.timeout(
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError
​
During handling of the above exception, another exception occurred:
​
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 678, in _watchdog_loop
    self.connection_lost(e)
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 688, in connection_lost
    self.listener_event("connection_lost", exc)
  File "/usr/local/lib/python3.12/site-packages/zigpy/util.py", line 50, in listener_event
    for listener, include_context in self._listeners.values():
RuntimeError: dictionary changed size during iteration
```

https://github.com/zigpy/zigpy/pull/1374 should suffer from the same issue. It seems like we attach/detach listeners inside of another listener, which should be supported.

MattWestb, 2024-05-25:
EZSP 6.10.X on Billy is also getting it but only the last part:
```
Logger: bellows.ezsp
Source: /usr/local/lib/python3.12/site-packages/bellows/ezsp/__init__.py:496
First occurred: 19:20:38 (1 occurrences)
Last logged: 19:20:38

Exception running handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bellows/ezsp/__init__.py", line 494, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.12/site-packages/bellows/zigbee/application.py", line 626, in ezsp_callback_handler
    self.connection_lost(args[0])
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 688, in connection_lost
    self.listener_event("connection_lost", exc)
  File "/usr/local/lib/python3.12/site-packages/zigpy/util.py", line 50, in listener_event
    for listener, include_context in self._listeners.values():
RuntimeError: 
 ```
Its my production system and its looks coming with some hours so it was one pain updating IKEA controllers that needing over 2 hours for doing the OTA.

TheJulianJES, 2024-05-25:
You seem to be having coordinator disconnects though. Are they expected?

MattWestb, 2024-05-25:
Yes i knowing but i dont think its the coordinator that is doing the time out its somthing lse in HA that is locking the thread to long so EZSP is timing out.
Most of the time UpNp and DLNA / cromecast is also having problems.
Not easy to knowing what was coming first the chicken or the egg as normal in the technical world.

PR #1406 (2024-06-05): Handle listeners that unregister themselves
Fixes #1404, fixes #1393.

Related: https://github.com/home-assistant/core/issues/116862
closed 2024-06-05 by puddly