zigpy/zigpy #1394: Updating HA to core 2024.5.0 crashed Silvercrest ZHA connecting

jayjayvm, 2024-05-02:
After updating to core 2024.5.0, My ZHA netwerk was vroken, no way to get the Silevercrest router starting again, heaps of debug info, some with fail most with succes, but it ends with ZHA trying to configure in device the router. All the tries end with this:

`2024-05-02 09:17:26.395 DEBUG (MainThread) [zigpy.application] Feeding watchdog
2024-05-02 09:17:26.396 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2024-05-02 09:17:26.397 DEBUG (MainThread) [bellows.uart] Sending: b'7d3162215754db2ea27e'
2024-05-02 09:17:26.450 DEBUG (MainThread) [bellows.uart] Data frame: b'1262a15754db15b259944a25aa5592499c4e27abedce678bfdc66389fc7e3fa7ebcdde6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a25190ceb17e'
2024-05-02 09:17:26.451 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2024-05-02 09:17:26.453 WARNING (MainThread) [bellows.ezsp.protocol] Failed to parse frame readCounters: b'0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
2024-05-02 09:17:26.454 DEBUG (MainThread) [bellows.uart] Connection lost: ValueError('Data is too short to contain 2 bytes')
2024-05-02 09:17:26.455 ERROR (MainThread) [bellows.uart] Lost serial connection: ValueError('Data is too short to contain 2 bytes')
2024-05-02 09:17:26.455 DEBUG (MainThread) [bellows.ezsp] socket://192.168.0.165:8888 connection lost unexpectedly: Data is too short to contain 2 bytes
2024-05-02 09:17:26.455 INFO (MainThread) [bellows.ezsp] NCP entered failed state. No application handler registered, ignoring...
`

Lost my whole netwerk to trying to migrate after many fails, hope this can be solved.

puddly, 2024-05-02:
Fixed by https://github.com/zigpy/bellows/pull/623

closed 2024-05-02 by puddly

jayjayvm, 2024-05-02:
Problem is, this solution is for people who know what they are doing in code, not for the average user. I have no clue what to do with it, and probably so many do not know. Or is it a fix that will be rolled out to a patched core now, very quickly.

puddly, 2024-05-02:
You've opened your issue in a dependency package of the ZHA integration. Follow https://github.com/home-assistant/core/issues/116575 if you're using HA.