zigpy/zigpy #1395: How to configure OTA for local image files？

hwzolin, 2024-05-06:
Hello, if I want to OTA the device through a local image file on ZHA for testing purposes, how should I configure it? Can you provide the latest configuration solution?
I found two methods on the Internet: otau_directory (link: https://community.home-assistant.io/t/zha-ota-firmware-update-upgrade/488960) and advanced_ota_dir (link: https://github. com/zigpy/zigpy/pull/1340).
Configured through these two methods, but in the end it prompts "No OTA image is available".

MattWestb, 2024-05-06:
I have the same problem trying updating IKEA devices with OTA files from TF feed locally and its not working at all so moving some devices to TF for updating and then back to ZHA (was working for some long time ago).

hwzolin, 2024-05-06:
@MattWestb 
Yes, although I don’t know which version this problem started from, but since I upgraded to a newer version, the local OTA cannot be used.

MattWestb, 2024-05-07:
Ping @puddly  !!!
Looks like somthing is being broken after some code changes then local and IKEA test (putting the normal TF URL in it i only getting warnings its desperate) is not working then trying updating devices with TF OTA files that is not in the DG feed like the new sensors and remote updates.

Thanks in advance our PUDDLY !!!!

puddly, 2024-05-07:
@hwzolin You would use this:

```yaml
zigpy:
  ota:
    advanced_ota_dir: /path/to/folder/with/ota/files
    allow_advanced_ota_dir: I understand I can *destroy* my devices by enabling OTA
      updates from files. Some OTA updates can be mistakenly applied to the
      wrong device, breaking it. I am consciously using this at my own risk.
```

I would, however, prefer https://github.com/zigpy/zigpy/pull/1398 once it is merged.

closed 2024-05-07 by puddly

MattWestb, 2024-05-07:
looks working for me after little "trixing" and putting this as OTA config in HA:
```
    ota:
      allow_advanced_ota_dir: I understand I can *destroy* my devices by enabling OTA updates from files. Some OTA updates can be mistakenly applied to the wrong device, breaking it. I am consciously using this at my own risk.
      advanced_ota_dir: /config/zigpy_ota
      ikea_provider: true
```
The long line shall not being broken up then its not being valid.