zigpy/zigpy #1404: Task exception was never retrieved / dictionary changed size during iteration

rrooggiieerr, 2024-05-24:
I'm getting quit a lot of exceptions which seem to cascade from zigpy_deconz to zigpy. Not sure which repository I should use for this issue so I decided to put it here

```
2024-05-24 09:47:29.011 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy_deconz/api.py", line 589, in _command
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 673, in _watchdog_loop
    await self.watchdog_feed()
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 655, in watchdog_feed
    await self._watchdog_feed()
  File "/usr/local/lib/python3.12/site-packages/zigpy_deconz/zigbee/application.py", line 87, in _watchdog_feed
    await self._api.write_parameter(
  File "/usr/local/lib/python3.12/site-packages/zigpy_deconz/api.py", line 849, in write_parameter
    await self.send_command(
  File "/usr/local/lib/python3.12/site-packages/zigpy_deconz/api.py", line 508, in send_command
    return await self._command(cmd, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy_deconz/api.py", line 588, in _command
    async with asyncio_timeout(COMMAND_TIMEOUT):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 678, in _watchdog_loop
    self.connection_lost(e)
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 688, in connection_lost
    self.listener_event("connection_lost", exc)
  File "/usr/local/lib/python3.12/site-packages/zigpy/util.py", line 50, in listener_event
    for listener, include_context in self._listeners.values():
RuntimeError: dictionary changed size during iteration

```

MattWestb, 2024-05-25:
One open issue looks having the same problem: https://github.com/home-assistant/core/issues/116862

TheJulianJES, 2024-05-29:
Duplicate of: https://github.com/zigpy/zigpy/issues/1393

PR #1406 (2024-06-05): Handle listeners that unregister themselves
Fixes #1404, fixes #1393.

Related: https://github.com/home-assistant/core/issues/116862
closed 2024-06-05 by puddly