zigpy/zigpy #1411: Network diagnostic senors needed !!!

MattWestb, 2024-06-10:
Some radios have counters for some network statistics but is not exposed to upper system.
The reason way need his being implanted is that i was having one very interesting problem im my production system.
Was testing binding one Vallhorn (the new IKEA motion sensor) with group and device binding with one tuya switch and was having problem with on wit off time not working at all but  normal On/Off was OK from ZHA GUI.
After sniffing it was very easy: tuya have not implanting On With Off time in this devices (as also the power mongering IC is not mounted on the PCB but its implemented in the quirk falsely).
Then giving up i was deleting the switch from the production network and all was looking OK.
Today i was having problem with one Parasoll (IKEA window / door sensor) and i was not knowing in with system its belonging so was sniffing both test networks and production.
Was seen that **all routers** was replaying to network allocation request breasted to them with one TELink IEEE with broadcast (and **deep 25 to 35 levels**) **every 4 seconds** = broadcast storm in 15.4 network layer.
After little filtering i was founding it was the Valhorn (Silabs IEEE) that was missing one bound device and looking for it "in one very nice way" by broadcasting network allocation request to all routers that was replaying with Success and `0` (= not in the neighbor table) with broadcast to all routers in the network.
Its not possible doing unbound in ZHA then the device is deleted in the system. But open the network and resetting it it was configured and working well without spamming the network.

The problem is its not possible getting the knowledge that the network have being blocked of broadcast storms so normal users cant fixing it.

Therefor the system needs counters that can being  nicely exposed in the upper system and also setting threshold if they is running amok like broadcast storms or link problems with reseeding in some network links that is degraded but cant being routed around of the mesh.

Feel free converting this issue to one discussion if like bu i think its one very importing functionality that is missing in the system for the moment.

puddly, 2024-06-10:
All EZSP counters are currently exposed as diagnostic entities in Home Assistant:
[image]

MattWestb, 2024-06-10:
One sniff from one normal "find one lost device":
```
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x99c8
    Frame Control Field: 0x8841, Frame Type: Data, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 6
    Destination PAN: 0x1a62
    Destination: 0xffff
    Source: 0x99c8
    [Extended Source: SiliconLabor_ff:fe:b8:35:56 (70:ac:08:ff:fe:b8:35:56)]
    [Origin: 28]
    TI CC24xx-format metadata: FCS OK
ZigBee Network Layer Data, Dst: Broadcast, Src: 0x59f5
    Frame Control Field: 0x1208, Frame Type: Data, Discover Route: Suppress, Security, Extended Source Data
    Destination: 0xfffd
    Source: 0x59f5
    Radius: 29
    Sequence Number: 152
    Extended Source: SiliconLabor_ff:fe:87:9c:b7 (5c:c7:c1:ff:fe:87:9c:b7)
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
    Frame Control Field: Data (0x08)
    Destination Endpoint: 0
    Network Address Request (Cluster ID: 0x0000)
    Profile: ZigBee Device Profile (0x0000)
    Source Endpoint: 0
    Counter: 173
ZigBee Device Profile, Network Address Request, Address: SiliconLabor_ff:fe:80:94:a0
    Sequence Number: 201
    Extended Address: SiliconLabor_ff:fe:80:94:a0 (14:b4:57:ff:fe:80:94:a0)
    Request Type: Single Device Response (0)
    Index: 0
```

Its not possible see if its one broadcast storm in the 15.4 layer then its need being how many unique broadcast is being seen by the coordinator / seconds or by the time inteval.

closed 2024-12-14 by github-actions[bot]