zigpy/zigpy #1428: Ikea Tradifi Update Brick the light

diegargon, 2024-07-25:
Zha update brick two new bulb i buy recently due update

The bulbs (shipped firmware)

TRADFRI bulb E27 WS globe 1055lm
Firmware: 0x03000010

Update to: Firmware: 0x03000022

I have old ones (that update seems not appears)

TRADFRIbulbE27WSglobeopal1055lm
Firmware: 0x00010012

puddly, 2024-07-25:
We disable IKEA updates by default and rely on the manufacturer to not push bad updates. I'll see what can be done though.

MattWestb, 2024-07-26:
@puddly Dont  disabling it then IKEA have released one fixed firmware  its in the reseal notes https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html (and reddit post from IKEA) and its some more updates for gen  3 light released last days that is working OK.

If have getting broke devices goo back to IKEA and return / change them it shall not being any problems at all.

(its not one test feed that was braking devices for some years ago iits the production feed = garantie is valid  then TF and DG is doing the same)

puddly, 2024-07-26:
@MattWestb  I'm seeing in the Reddit thread a lot of people mention that `3.0.22` bricked it. I think we can just disable updates for this specific device until this is confirmed fixed.

In the future, I'm hoping to set up https://github.com/zigpy/ota and have all OTA updates from all manufacturers committed to the repository as pull requests. Beta users can install them immediately. Stable users only get the update after a week or two.

puddly, 2024-07-26:
I'm actually not able to find `0x03000022` (`50331682`) at all in the current DIRIGERA JSON:

```console
$ curl -k -L 'https://fw.ota.homesmart.ikea.com/DIRIGERA/version_info.json' | jq
```

<details>
<summary>Output</summary>

```json
[
  {
    "fw_image_type": 10242,
    "fw_type": 2,
    "fw_sha3_256": "e68e61bd57291e0b6358242e72ee2dfe098cb8b769f572b5b8f8e7a34dbcfaca",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/mgm210l-light-cws-cv-rgbw_release_prod_v268572245_3ae78af7-14fd-44df-bca2-6d366f2e9d02.ota"
  },
  {
    "fw_update_prio": 5,
    "fw_filesize": 262600739,
    "fw_type": 3,
    "fw_hotfix_version": 0,
    "fw_major_version": 2,
    "fw_binary_checksum": "7c5fbb0ada1f6f01b516129604e70bf09cd4fdefc2e09dac58eb05e722c6781d8baafb81287933e2149b4003dcd52b83b6265436cbd40625c9cff58080cb94ad",
    "fw_minor_version": 588,
    "fw_sha3_256": "f73d1ab3ff9677d9cee8d3b0da81aec708213367908eb4d2d09af0d26244875a",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/DIRIGERA_release_prod_v2.588.0_24d89869-0b44-4a58-b29f-c5d3127d3931.raucb"
  },
  {
    "fw_image_type": 6456,
    "fw_type": 2,
    "fw_sha3_256": "ad623ed146226dc17b76b71a7a8414c6e550cfe29513580bd4d9269bc7578b2b",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/motionsensor-lds-mg21a_release_prod_v16777316_3a00064a-cc29-4aac-bdb6-c4fa1fb445d5.ota"
  },
  {
    "fw_image_type": 4545,
    "fw_type": 2,
    "fw_sha3_256": "d36803d6dea5f86bc3aea3b3ff7fd46d5d2187b61209cf68ceea2474e85a5f48",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/tradfri-controller_release_prod_v604241925_abef4451-762a-4ef9-8c11-dfd88c3e98f7.ota"
  },
  {
    "fw_image_type": 4554,
    "fw_type": 2,
    "fw_sha3_256": "3c8cd9a5eee7e1b35a187e84a91db5d2bc91fd98c27f349548ae211eb233720d",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/tradfri-dimmer_release_prod_v604241925_ecbb4451-ce85-4e6c-ab9f-e7ce32cd0c1e.ota"
  },
  {
    "fw_image_type": 4548,
    "fw_type": 2,
    "fw_sha3_256": "81edaa25a0e5c31a594b41551f1ebdede6c5f94b58e2dfe5462915fc0d4fc637",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/tradfri-motion-sensor_release_prod_v604241925_e006dcd0-d37e-49db-9a61-6575cb5c44d3.ota"
  },
  {
    "fw_image_type": 16897,
    "fw_type": 2,
    "fw_sha3_256": "87b25a4ca55b1920d50defffd1bd459f0c90a9af0adc6e51609ebd384d6e78ac",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/tradfri-driver-lp_release_prod_v587757105_b28cda41-22d2-446b-b3bf-5ca11d866719.ota"
  },
  {
    "fw_image_type": 4550,
    "fw_type": 2,
    "fw_sha3_256": "ce7b04ec3ccaa18e0b8b6a085a61c9526e3bacdc6b3513352ac62d2689b066eb",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/tradfri-shortcut-button_release_prod_v604241926_56f5d8d1-78b1-4088-afc1-05d3b7e3314b.ota"
  },
  {
    "fw_image_type": 8710,
    "fw_type": 2,
    "fw_sha3_256": "4992557dad942e9b56f7171c34186005e1b329a57cc23c9c140addfd153eedd9",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/zingo-ws-soc_release_prod_v50331664_500d55d8-d55d-46f0-9afd-68d88ec514dd.ota"
  },
[... 209 lines cut ...]
  },
  {
    "fw_image_type": 10241,
    "fw_type": 2,
    "fw_sha3_256": "e718abba1958fe7e00297a1c143033d8505b26b67d2668faf08bc137cc27ba00",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/tradfri-bulb-cws-zll_release_prod_v587753009_ec8b1193-0fa2-440e-b921-2412a8688b74.ota"
  }
]
```
</details>

The only similar image is this one:

```json
  {
    "fw_image_type": 8704,
    "fw_type": 2,
    "fw_sha3_256": "b3fec5e75bb4c529b84bd2e92a40146d992672d6afe943aef8349ccfaa0695cd",
    "fw_binary_url": "https://fw.ota.homesmart.ikea.com/files/zingo-ws_release_prod_v50331681_969bee21-eca6-4f10-a4c0-9685c0cd5d52.ota"
  },
```

And it is version `50331681`, not `50331682`. Seems to be `3.0.21`? Maybe IKEA pulled 3.0.22.

Just to be safe, is `image_type: 8704` the only affected bulb?

MattWestb, 2024-07-26:
Its only in TF feed like some other not certificated (or at least published) firmware.
Others is only DG so IKEA is messing  around little for the moment.

From my download script it  shall being `zingo_ww-0x2100-1.0.42-prod.ota.ota.signed`.

MattWestb, 2024-07-26:
The `TRADFRIbulbE27WSglobeopal1055lm` is Gen3 but have no resent update and is not being broken.
From my test device:

Device info
TRADFRIbulbE27WSglobeopal1055lm
by IKEA of Sweden
Connected via [IKEA Billy EZSP 6.10.7.0](http://192.168.2.66:8123/config/devices/device/9b13acab058111ebb300cf2132e87262)
Firmware: 0x00010012

The Gen1/2  firmware from TF feed for Puddly:
[zingo_ws_soc-0x2206-3.0.22-prod.ota.ota.zip](https://github.com/user-attachments/files/16393962/zingo_ws_soc-0x2206-3.0.22-prod.ota.ota.zip)

puddly, 2024-07-26:
Thanks! So will blocking `8710` for IKEA be enough then to prevent updates for this specific device? Or is there a second variant (`zingo-ws` has two types, the other being `8704`)?

MattWestb, 2024-07-26:
The `zingo_ws | 0x2200 8704` is OK was getting  devices 4 updated early this morning (TF feed).
(just sniffed one new request for verification and its the right working on for the image) 

I dont have the image in version 3.0.21 but that  shall being the braking one and its update in TF to 3.0.22 so shall being OK but i dont know the DG feed is updating so  you need looking on it.

TheJulianJES, 2024-07-26:
To make sure I understand everything correctly, @MattWestb:
1. The broken update that bricked devices was already pulled by IKEA, right? (yes/no)
2. The broken update was replaced by a newer update that no longer bricks bulbs? (yes/no)
3. Already bricked bulbs have to be exchanged at IKEA. You cannot update them now. (yes/no)

If the assumptions are correct, is there any reason to still block firmware updates for that device then?

puddly, 2024-07-26:
Here's what I get for all three feeds:

```python
In [1]: from zigpy.ota.providers import Tradfri

In [2]: dirigera = await Tradfri('https://fw.ota.homesmart.ikea.com/DIRIGERA/version_info.json').load_index()

In [3]: ikea_old = await Tradfri('http://fw.ota.homesmart.ikea.net/feed/version_info.json').load_index()

In [4]: ikea_old_test = await Tradfri('https://fw.test.ota.homesmart.ikea.net/feed/version_info.json').load_index()

In [5]: [i for i in dirigera if i.image_type == 8704]
Out[5]: [IkeaRemoteOtaImageMetadata(file_version=50331681, manufacturer_id=4476, image_type=8704, checksum='sha3-256:b3fec5e75bb4c529b84bd2e92a40146d992672d6afe943aef8349ccfaa0695cd', file_size=None, manufacturer_names=(), model_names=(), changelog=None, release_notes=None, min_hardware_version=None, max_hardware_version=None, min_current_file_version=None, max_current_file_version=None, specificity=None, source='IKEA (DIRIGERA)', url='https://fw.ota.homesmart.ikea.com/files/zingo-ws_release_prod_v50331681_969bee21-eca6-4f10-a4c0-9685c0cd5d52.ota', ssl_ctx=None)]

In [6]: [i for i in ikea_old if i.image_type == 8704]
Out[6]: [SignedIkeaRemoteOtaImageMetadata(file_version=50331681, manufacturer_id=4476, image_type=8704, checksum=None, file_size=307758, manufacturer_names=(), model_names=(), changelog=None, release_notes=None, min_hardware_version=None, max_hardware_version=None, min_current_file_version=None, max_current_file_version=None, specificity=None, source='IKEA (TRÅDFRI)', url='https://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/zingo_ws-0x2200-3.0.21-prod.ota.ota.signed', ssl_ctx=None)]

In [7]: [i for i in ikea_old_test if i.image_type == 8704]
Out[7]: []
```

So it looks like `3.0.21` is the only firmware available. It *sounds* on Reddit like the `3.0.21` update was the bad one? Or maybe it's the last firmware version that could be read from the device, so it was the `3.0.22` update that broke these bulbs?

puddly, 2024-07-26:
@diegargon Can you post your ZHA OTA config? Are you using the default IKEA provider (DIRIGERA) or did you manually use the old test feed URL?

MattWestb, 2024-07-26:
From time machine:

15/7: http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/zingo_ws_soc-0x2206-3.0.21-prod.ota.ota.signed"

25/7: http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/zingo_ws_soc-0x2206-3.0.22-prod.ota.ota.signed"

So IKEA have putting out one update after the  request they was doing on reddit and also doing the release notes.

My TF script is getting the 3.0.22 for  5 minutes ago so little strange that you is getting the 3.0.21 puddly.

MattWestb, 2024-07-26:
> The last 64 bytes of the flash data in the firmwares themselves differ between the DIRIGERA and the test feed firmwares, I'm wondering if the test feed is broken?

Perhaps have use different signing key or padding the end before the CRC and / or  the signing.

Can  you finding 0x2206 / 8710 in Dirigera feed  ??
If not then  TF is working OK then its coming from it for my updated GU10 WS (or can being DG if its also the).

MattWestb, 2024-07-26:
@puddly have you restarting HA  for getting the update feed then its looks strange you is getting one old version from it ?

puddly, 2024-07-26:
Strange. The code I ran above was outside of Home Assistant, it was just zigpy and shouldn't be caching anything. I'll try again later tonight and change the URLs to invalidate the cache (adding `?somerandomtext` might work)

puddly, 2024-07-26:
Ahh, we're looking at different data. I was looking at 8704, not 8710! Here is all of the data:

| Image Type |   Feed   |    Version   |
| --------------- | -------- | ------------ |
| 8710            | DIRIGERA | `0x03000010` |
| 8710            | Old OTA  | `0x03000022` |
| 8704            | DIGIGERA | `0x03000021` |
| 8704            | Old OTA  | `0x03000021` |

So for 8710, DIRIGERA is still on 3.0.10 but the old OTA feed is on 3.0.22, the broken version? 8704 is on 3.0.21 for both feeds.

I think 8710 is the only affected device, 8704 isn't affected. And it looks like the old OTA feed is still pushing the bad image...

MattWestb, 2024-07-26:
The` zingo_ws | 0x2200 8704` is OK.
0x2206 8710 `zingo_ws_soc-0x2206-3.0.22-prod.ota.ota.signed` from TF shall being the fixed version (was  reading little more on reddit and possible recover if not re-power  the device and if done = hard braked device) and Dirigera  is being hold back = OK then its costing little braking some hundred thousand devices.

So IKEA have all feed fixed or held back (and very likely not rolling out the braked version in DG feed).

MattWestb, 2024-07-26:
No need  blocking  `zingo_ws` | 0x2200 / 8704 its working OK.

[image]

I have 4 in production and 2 in RCP test updated and working OK

diegargon, 2024-07-26:
> @diegargon Can you post your ZHA OTA config? Are you using the default IKEA provider (DIRIGERA) or did you manually use the old URL?
> 
> The last 64 bytes of the flash data in the firmwares themselves differ between the DIRIGERA and the test feed firmwares, I'm wondering if the test feed is broken? I noticed only two reports from ZHA users overall, everything else is coming from Z2M. We don't use the test feed unless you manually configure it.

I not do nothing special, just have that in my configuration file.

zha:
  database_path: /config/zigbee.db 
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota
      ikea_provider: true
     sonoff_provider: true

When i connect the 3 new lights the update appears in HA,  22 version not 21,  i click update  two of them and they broke

The survive light:

TRADFRI bulb E27 WS globe 1055lm
por IKEA of Sweden
Firmware: 0x03000010
Zigbee info
IEEE: b0:c7:de:ff:fe:64:dd:58
Nwk: 0x8667
Device Type: Router
LQI: 168
RSSI: -58
Visto por última vez: 2024-07-27T00:29:28
Fuente de alimentación: Mains

PR #1430 (2024-07-29): Block updates for IKEA image types 8704 and 8710
Fixes #1428.

We'll re-enable them whenever IKEA fixes this problem. I'd rather not take the risk of figuring out which exact devices are affected.
closed 2024-07-29 by puddly