zigpy/zigpy #1445: z2m_local not working? missing 1 required positional argument: 'index_file'

eiten, 2024-08-13:
HA 2024.08.1
I got the following `configuration.yaml`:
`zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: z2m_local
          path: /config/test/index.json`

`/config/test/index.json` looks like this:
`[
    {
        "fileVersion": 2,
        "fileSize": 314340,
        "manufacturerCode": 4735,
        "imageType": 321,
        "sha512": "0c1e3a5c1b374128731e93ffe8889ef6f3a5103b9b0e61f3ff761294f0d5185f53550da9d0a341fabd7e9234ba8cbdeecddfb3bf42c8f33a496cb0a3bfb44109",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Nordic%20Semiconductors/127F-0141-00000002-light_switch.zigbee",
        "path": "/config/test/127F-0141-00000002-light_switch.zigbee"
    }
]`

I get an error about `missing 1 required positional argument: 'index_file'`:
`Logger: homeassistant.config_entries
Quelle: config_entries.py:604
Erstmals aufgetreten: 12:02:44 (1 Vorkommnisse)
Zuletzt protokolliert: 12:02:44

Error setting up entry slae.sh cc2652rb stick - slaesh's iot stuff - slae.sh cc2652rb stick - slaesh's iot stuff for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 132, in async_setup_entry
    async with radio_mgr.connect_zigpy_app() as app:
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 182, in connect_zigpy_app
    app = await self.radio_type.controller.new(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 250, in new
    app = cls(config)
          ^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy_znp/zigbee/application.py", line 81, in __init__
    super().__init__(config=config)
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 67, in __init__
    self._config = self.SCHEMA(config)
                   ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 549, in validate_dict
    return base_validate(path, data.items(), out)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 330, in validate_mapping
    cval = cvalue(key_path, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 549, in validate_dict
    return base_validate(path, data.items(), out)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 330, in validate_mapping
    cval = cvalue(key_path, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 590, in validate_sequence
    cval = validate(index_path, value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/config/validators.py", line 144, in cv_ota_provider
    return provider_cls(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^
TypeError: LocalZ2MProvider.__init__() missing 1 required positional argument: 'index_file'`

puddly, 2024-08-13:
Try replacing `path` with `index_file` and point it to the `index.json` file.

eiten, 2024-08-14:
You're welcome! I'm glad it's a bug, I started to doubt myself :). The thing with `index_file` i already tried myself, sorry for not mentioning it.

PR #1447 (2024-08-14): Fix local Z2M provider
Fixes #1445.