zigpy/zigpy #1454: Array object is not iterable

prairiesnpr, 2024-08-16:
Parsing a ReadAttributesResp from the StateText attribute of a MultiStateInput cluster fails with the following error.

```
2024-08-13 21:03:47.475 DEBUG (MainThread) [zigpy.zcl] [0x4F4E:1:0x0012] Decoded ZCL frame: MultistateInput:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x000E, status=<Status.SUCCESS: 0>, value=TypeValue(type=Array, value=Array(type=AnonymousLVList, value=['Full', 'Low', 'Empty'])))])
2024-08-13 21:03:47.476 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [545072371536] Error handling message: Unknown error (unknown_error) ---- from 172.0.0.1 (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0)
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/websocket_api/decorators.py", line 28, in _handle_async_response
    await func(hass, connection, msg)
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/zha/websocket_api.py", line 861, in websocket_read_zigbee_cluster_attributes
    success, failure = await cluster.read_attributes(
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/zigpy/zcl/__init__.py", line 538, in read_attributes
    value = self.attributes[record.attrid].type(record.value.value)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'Array' object is not iterable```

puddly, 2024-08-16:
Can you include the entire debug log (or at least 20 lines above this)? It's easier to test types if you have the actual Zigbee packet structure.

prairiesnpr, 2024-08-16:
This is with the changes in https://github.com/zigpy/zigpy/pull/1455, but packet is the same.

```
2024-08-16 10:44:22.889 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 8, 16, 14, 44, 22, 889629, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4F4E), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=18, data=Serialized[b'X\n\x01\x0e\x00\x00HB\x03\x00\x04Full\x03Low\x05Empty'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=112, rssi=None)
2024-08-16 10:44:22.890 DEBUG (MainThread) [zigpy.zcl] [0x4F4E:1:0x0012] Received ZCL frame: b'X\n\x01\x0e\x00\x00HB\x03\x00\x04Full\x03Low\x05Empty'
2024-08-16 10:44:22.891 DEBUG (MainThread) [zigpy.zcl] [0x4F4E:1:0x0012] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x58>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=2, *is_cluster=False, *is_general=True), tsn=10, command_id=1, *direction=<Direction.Server_to_Client: 1>)
2024-08-16 10:44:22.892 DEBUG (MainThread) [zigpy.zcl] [0x4F4E:1:0x0012] Decoded ZCL frame: MultistateInput:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x000E, status=<Status.SUCCESS: 0>, value=TypeValue(type=Array, value=Array(type=AnonymousLVList, value=['Full', 'Low', 'Empty'])))])
2024-08-16 10:44:22.893 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x4F4E:1:0x0012]: cluster_handler[multistate_input] attribute_updated - cluster[MultistateInput] attr[state_text] value[Array(type=AnonymousLVList, value=['Full', 'Low', 'Empty'])]
2024-08-16 10:44:22.893 DEBUG (MainThread) [zha] Emitting event cluster_handler_attribute_updated with data ClusterAttributeUpdatedEvent(attribute_id=14, attribute_name='state_text', attribute_value=Array(type=AnonymousLVList, value=['Full', 'Low', 'Empty']), cluster_handler_unique_id='00:13:a2:00:41:67:1f:d7:1:0x0012', cluster_id=18, event_type='cluster_handler_event', event='cluster_handler_attribute_updated') (1 listeners)

```

puddly, 2024-08-17:
Looks like the data format for attribute reads for arrays is different 😄
[image]

closed 2024-08-27 by puddly