zigpy/zigpy #1465: Disconnects every 30 mins or so from my coordinator

thefunkygibbon, 2024-09-04:
Using latest HA 2024.8.3 on a container but its been like this for at least a few days now so i'm unsure how many versions. 
SLZB-06M is my coordinator.   connected via ethernet.  no connectivity issues between them (have been pinging both constantly and there is no dropped packets or even anything which goes above a few ms latency)

noticing the zha restarting often (again).  

I think the below is likely relevant from the times where it last connected to my coordinator ?

2024-09-04 10:38:53.714 DEBUG (MainThread) [bellows.ash] No ACK received in 0.40s (attempt 1)
2024-09-04 10:38:53.714 DEBUG (MainThread) [bellows.ash] Changing ACK timeout from 0.40 to 0.80
2024-09-04 10:38:53.714 DEBUG (MainThread) [bellows.ash] Sending frame DataFrame(frm_num=2, re_tx=True, ack_num=2, ezsp_frame=b'\x11\x00\x01\xaa\x00\x03') + FLAG
2024-09-04 10:38:53.714 DEBUG (MainThread) [bellows.ash] Sending data  2a5321a9fe2a160a6f7e
2024-09-04 10:38:54.395 DEBUG (MainThread) [bellows.ash] Received data 1ac2020682af7e
2024-09-04 10:38:54.395 DEBUG (MainThread) [bellows.ash] Received cancel byte, clearing buffer
2024-09-04 10:38:54.395 DEBUG (MainThread) [bellows.ash] Received frame ErrorFrame(version=2, reset_code=<NcpResetCode.RESET_ASSERT: 6>)
2024-09-04 10:38:54.395 DEBUG (MainThread) [bellows.ash] NCP has entered failed state: NcpResetCode.RESET_ASSERT
2024-09-04 10:38:54.396 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2024-09-04 10:38:54.396 DEBUG (MainThread) [bellows.zigbee.application] Received _reset_controller_application frame with (<NcpResetCode.RESET_ASSERT: 6>,)
2024-09-04 10:38:54.396 DEBUG (MainThread) [zigpy.application] Connection to the radio has been lost: <NcpResetCode.RESET_ASSERT: 6>
2024-09-04 10:38:54.396 DEBUG (MainThread) [zha.application.gateway] Connection to the radio was lost: <NcpResetCode.RESET_ASSERT: 6>
2024-09-04 10:38:54.396 DEBUG (MainThread) [zha] Emitting event connection_lost with data ConnectionLostEvent(event_type='zha_gateway_message', event='connection_lost', exception=<NcpResetCode.RESET_ASSERT: 6>) (1 listeners)
2024-09-04 10:38:54.397 DEBUG (MainThread) [zha] (ZHAGatewayProxy) handling event protocol for event: ConnectionLostEvent(event_type='zha_gateway_message', event='connection_lost', exception=<NcpResetCode.RESET_ASSERT: 6>)
2024-09-04 10:38:54.397 DEBUG (MainThread) [homeassistant.components.zha.helpers] Connection to the radio was lost: ConnectionLostEvent(event_type='zha_gateway_message', event='connection_lost', exception=<NcpResetCode.RESET_ASSERT: 6>)
2024-09-04 10:38:54.397 DEBUG (MainThread) [zha.application.helpers] stopping global updater
2024-09-04 10:38:54.397 DEBUG (MainThread) [zha.application.helpers] global updater stopped
2024-09-04 10:38:54.397 DEBUG (MainThread) [zha.application.helpers] stopping device availability checker
2024-09-04 10:38:54.397 DEBUG (MainThread) [zha.application.helpers] device availability checker stopped
2024-09-04 10:38:54.399 DEBUG (MainThread) [zha.application.platforms] 00:17:88:01:10:44:67:6a-11: Cancelling task: <Task pending name='light_refresh_00:17:88:01:10:44:67:6a-11' coro=<periodic.<locals>.scheduler.<locals>.wrapper() running at /usr/local/lib/python3.12/site-packages/zha/decorators.py:75> wait_for=<Future pending cb=[Task.task_wakeup()]> cb=[set.remove()]>
2024-09-04 10:38:54.399 DEBUG (MainThread) [bellows.ash] NCP has entered into a failed state, not retrying
2024-09-04 10:38:54.399 DEBUG (MainThread) [bellows.uart] Connection lost: None
2024-09-04 10:38:54.400 DEBUG (MainThread) [bellows.uart] Closed serial connection
2024-09-04 10:38:54.401 WARNING (MainThread) [zigpy.application] Watchdog failure
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 665, in _watchdog_loop
    await self.watchdog_feed()
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 651, in watchdog_feed
    await self._watchdog_feed()
  File "/usr/local/lib/python3.12/site-packages/bellows/zigbee/application.py", line 903, in _watchdog_feed
    free_buffers = await self._get_free_buffers()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bellows/zigbee/application.py", line 921, in _get_free_buffers
    status, value = await self._ezsp.getValue(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bellows/ezsp/__init__.py", line 221, in _command
    return await command(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bellows/ezsp/protocol.py", line 75, in command
    await self._gw.send_data(data)
  File "/usr/local/lib/python3.12/site-packages/bellows/uart.py", line 57, in send_data
    await self._transport.send_data(data)
  File "/usr/local/lib/python3.12/site-packages/bellows/ash.py", line 680, in send_data
    await asyncio.shield(
  File "/usr/local/lib/python3.12/site-packages/bellows/ash.py", line 631, in _send_data_frame
    await ack_future
bellows.ash.NcpFailure
2024-09-04 10:38:54.405 DEBUG (MainThread) [zigpy.application] Connection to the radio has been lost: <NcpFailure(code=<NcpResetCode.RESET_ASSERT: 6>)>
2024-09-04 10:38:54.405 DEBUG (MainThread) [zha.application.gateway] Connection to the radio was lost: <NcpFailure(code=<NcpResetCode.RESET_ASSERT: 6>)>
2024-09-04 10:38:54.405 DEBUG (MainThread) [zha] Emitting event connection_lost with data ConnectionLostEvent(event_type='zha_gateway_message', event='connection_lost', exception=<NcpFailure(code=<NcpResetCode.RESET_ASSERT: 6>)>) (0 listeners)
2024-09-04 10:38:54.405 DEBUG (MainThread) [zigpy.application] Stopping watchdog loop

2024-09-04 10:57:56.061 DEBUG (MainThread) [bellows.ash] Received frame DataFrame(frm_num=5, re_tx=1, ack_num=5, ezsp_frame=b'\x93\x00\x01\xaa\x00\x03')
2024-09-04 10:57:56.061 DEBUG (MainThread) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=6) + FLAG
2024-09-04 10:57:56.061 DEBUG (MainThread) [bellows.ash] Sending data  8610be7e
2024-09-04 10:57:56.062 WARNING (MainThread) [bellows.ezsp.protocol] Failed to parse frame getValue: b'03'
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bellows/ezsp/protocol.py", line 108, in __call__
    result, data = t.deserialize_dict(data, rx_schema)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bellows/types/__init__.py", line 9, in deserialize_dict
    value, data = type_.deserialize(data)
                  ^^^^^^^^^^^^^^^^^^^^^^^
[... 24 lines cut ...]
    if extra_state_attributes := self.extra_state_attributes:
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/omada/sensor.py", line 904, in extra_state_attributes
    return self.entity_description.extra_attributes_fn(self.controller)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/omada/sensor.py", line 71, in controller_clients_extra_attributes_fn
    attributes["clients"] = sorted(clients, key=lambda x: ipaddress.IPv4Address(x["ip"]))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/omada/sensor.py", line 71, in <lambda>
    attributes["clients"] = sorted(clients, key=lambda x: ipaddress.IPv4Address(x["ip"]))
                                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/ipaddress.py", line 1319, in __init__
    self._ip = self._ip_int_from_string(addr_str)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/ipaddress.py", line 1206, in _ip_int_from_string
    raise AddressValueError("Expected 4 octets in %r" % ip_str)
ipaddress.AddressValueError: Expected 4 octets in 'None'

2024-09-04 10:59:10.944 DEBUG (MainThread) [bellows.ash] Received data 7d31adb1a97d312a15b65894a524ab5593499c84f37a10379874f0cf0b8bfce536a5ebc9de6f8e7d5e2f677e
2024-09-04 10:59:10.945 DEBUG (MainThread) [bellows.ash] Received frame DataFrame(frm_num=1, re_tx=0, ack_num=1, ezsp_frame=b'\xef\x90\x01E\x00\x00\x04\x01\x00\xef\x01\x01\x00\x01\x00\x00\xca\xd4\xd1\xfd\xf9\xff\xff\r\th\x02\x00\x9b\t\x02\x00\x04\x00\x00\x01\x81')
2024-09-04 10:59:10.945 DEBUG (MainThread) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=2) + FLAG

Also here is a full debug log taken from a couple of minute interval where the issue 100% occured. 

[zhalog1.txt](https://github.com/user-attachments/files/16869372/zhalog1.txt)

thefunkygibbon, 2024-09-11:
I changed firmware of my coordinator and it no longer is an issue.  it seems that the problem was something to do with it (although i'm not sure what)

dmulcahey, 2024-09-11:
So we can close this?

closed 2024-09-11 by thefunkygibbon