zigpy/zigpy #1468: Allow factory reset of Philips Hue Zigbee Light Link (ZLL) devices through TouchLink?

Hedda, 2024-09-10:
Feature request that not sure if belongs to Zigbee gateway library or [zigpy-cli](https://github.com/zigpy/zigpy-cli) as a tool, but think it should be added to ZHA integration.

Anyway, please consider adding support for factory resetting older Philips Hue Zigbee Light Link (ZLL) devices through Touchlink (like Zigbee2MQTT and the hue-thief script can do) as that would make it much more user-friendly to pair/join Philips Hue bulbs with older firmware that has previously been added to another Zigbee Gateway (like example the official Philips Hue Bridge) but then not been properly removed and reset. 

* https://github.com/vanviegen/hue-thief/

hue-thief (with a EZSP-based Zigbee adapter) is a Python script that can also be used factory reset Hue devices through Touchlink: 
_**Hue Thief will scan all Zigbee channels for ZLL-compatible bulbs that are associated with any Zigbee network. When a bulb is found, it will blink a couple of times, and the application will ask if you want to factory reset this bulb. (If you didn't see any blinking, you may be doing your neighbours a favour by choosing 'N' here. :-)) That's it. Your now factory clean bulbs should be discoverable through whatever means your bridge/software offers.**_

* https://www.zigbee2mqtt.io/guide/usage/touchlink.html#factory-reset-device

Demo video showing Zigbee2mqtt factory reset Hue bulb with Touchlink:

* https://www.youtube.com/watch?v=kcRj77YGyKk

Problem to solve:  Philip Hue devices with older firmware can be a real pain to factory reset if you do not have the official Philips Hue Bridge because they deviated from the official Zigbee specification, so older Philip Hue lightbulbs are only capable of being factory reset via Touchlink, and as such it is not possible to factory reset such older Philip Hue devices by just pressing a hardware reset-button or use the more common power-cycle reset method (i.e. simply turn the power off and on to the device in a certain order with correct timing a with most other Zigbee devices.

* https://www.zigbee2mqtt.io/guide/usage/touchlink.html

As it stands right now ZHA end-users have to jump through some hops before they can properly pair/join a previously used such Philips Hue lightbulb device to ZHA if they do not still have their official Philips Hue Bridge, and the current workarounds is to either use a Hue compatible remote or dimmer switch product or hue-thief reset the bulb/light via Touchlink. Only after a such factory reset can such older Philips Hue bulbs easily join ZHA (or other compatible Zigbee gateway).

Those workarounds are covered in ZHA docs under troubleshooting here:

* https://www.home-assistant.io/integrations/zha#add-philips-hue-bulbs-that-have-previously-been-added-to-another-bridge

Having this feature would especially make it much easier for new users migrate from a Philips Hue Bridge to the ZHA integration:

* https://community.home-assistant.io/t/migrating-from-a-philips-hue-bridge-to-the-zha-zigbee-home-automation-integration/620862

PS: Off-topic but you might also want to consider adding support for TouchLink commissioning which can allow users to commission and pair a TouchLink capable Zigbee device by simply putting it in very close physical proximity to the Zigbee Coordinator and selecting "TouchLink commissioning" to start exchanging keys, which is known “Touchlinking" in the Zigbee Light Link (ZLL) specification, (downside to having a such feature is that it is at risk of a security attack during the touchlink commissioning procedure):

* https://www.silabs.com/documents/public/user-guides/ug103-09-fundamentals-zll.pdf

johnneerdael, 2025-01-08:
https://github.com/johnneerdael/hue-thief-skyconnect

I've made a heavily modified version specifically for the skyconnect intended to run on win (or Mac using a virtualized ubuntu for instance (done using UTM here on a M3)

Hedda, 2025-01-08:
> https://github.com/johnneerdael/hue-thief-skyconnect
> 
> I've made a heavily modified version specifically for the skyconnect

Any chance you consider looking into porting it as a feature inside zigpy-cli or zha library and ZHA integration component for UI?

https://github.com/zigpy/zigpy-cli

https://github.com/zigpy/zha

https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha

Alternativly as a new feature for the zha-toolkit if the zigpy developers do not want that code inside a zigpy project?

https://github.com/mdeweerd/zha-toolkit

https://community.home-assistant.io/t/zha-toolkit-a-big-set-of-zigbee-commands-on-top-of-zha-zigpy/373346

johnneerdael, 2025-01-10:
> Any chance you consider looking into porting it as a feature inside zigpy-cli or zha library and ZHA integration component for UI?
>
> https://github.com/zigpy/zigpy-cli
>
> https://github.com/zigpy/zha [...]

Honestly I found most reliable way for Philips bulbs to do following power cycle till it resets: 2s off, 8s on, 2s off, 5s on and repeat a few times, worked every time. On traditional devices you just power cycle 6 times in a row quickly.

puddly, 2025-06-29:
This is tricky to implement and will require a ton of changes to zigpy, unfortunately. We effectively have to disable the radio for the entire duration of the scan and will need a new API to switch the adapter into inter-PAN mode. This is likely something we can add to zigpy-cli, however, and possibly to ZHA as an action once the API is stabilized.

We added the TouchLink clusters to zigpy at some point in the past (https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/lightlink.py) but the problem here is the radio API. I think we can extend the `ZigbeePacket` type to include optional `pan_id` and `channel` fields and then have a context manager like this?

```python
async with app.interpan_mode():
    await app.set_interpan_channel(11)
    await app.send_interpan_packet(...)
```

For actually receiving packets, however, `app.wait_for_response` will need to be amended to work with inter-PAN packets, since we assume all communication is done device-to-device with known devices on the current network, not ephemeral nodes like this.

A-Mundanilkunathil, 2025-07-12:
## Touchlink Reset Implementation Question

Hi @Puddly,

I wanted to check if I'm on the right track with implementing **Touchlink reset** in `zigpy` and exposing it via **ZHA**.

We're working on this as part of **Code Day** and I'm collaborating with @janeswingler and @justin2186 on this implementation.

---

## Current Setup

I've cloned and am working locally with:
- `home-assistant/core`
- `zigpy/zha`
- `zigpy/zigpy`
- `zigpy/zigpy-cli`
- `zigpy/bellows` and `zigpy/znp` (to explore both radio paths)

**Hardware:**
- SONOFF ZBDongle-P (CC2652P)
- Home Assistant Connect ZBT-1 (EFR32MG21)

**Test devices:**
- GE Link BR30
- IKEA TRÅDFRI E26

*(Both should support Touchlink reset)*

---

## Implementation Plan

### 1. Add inter-PAN mode support to zigpy core:
- [ ] Add an `interpan_mode()` context manager in `zigpy.application.ControllerApplication`
- [ ] Add abstract `_enter_interpan_mode()` and `_exit_interpan_mode()` methods to the radio interface
- [ ] Add Touchlink service methods (`touchlink_scan()`, `touchlink_reset_device()`) to `ControllerApplication`

### 2. Add the necessary inter-PAN mode controls to the radio libraries (bellows, znp):
- [ ] Implement `_enter_interpan_mode()` and `_exit_interpan_mode()` in both radio libraries
- [ ] Handle channel switching and PAN ID management for inter-PAN operations

### 3. Use zigpy-cli to prototype the command and verify the hardware works

### 4. Once working, wire it up as a ZHA service (e.g. `zha.touchlink_reset`) in Home Assistant

---

## Additional Considerations

Since `zigpy.zcl.clusters.lightlink` already exists, I was also wondering:

> Should I be thinking about implementing a more formal **"Touchlink service"** class/module that uses those LightLink cluster commands, or is there a preferred way you'd envision this being integrated into the Zigpy architecture?

---

## Questions

- **Does this approach seem sound**, or am I misunderstanding any part of how inter-PAN packets should flow through the stack?
- **What's your preferred approach** for handling cases where the radio doesn't support inter-PAN mode or Touchlink operations?

Any guidance, validation, or red flags you can point out would be greatly appreciated — thank you again for all your work on this ecosystem!

Hedda, 2025-08-01:
While this is all way too low-level stuff for me I am still wondering if it might be a good idea to look how Touchlink is handled in the [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman) which is used by the [zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt) application as they have implemented factory reset via Touchlink feature? Maybe for reference see:

https://github.com/search?q=repo%3AKoenkk%2Fzigbee-herdsman%20touchlink&type=code

and

https://github.com/search?q=repo%3AKoenkk%2Fzigbee2mqtt%20touchlink&type=code

Hedda, 2025-11-21:
FYI, for reference, it looks like Zigbee2MQTT's developers started working on something related for their zigbee-herdsman library:

* https://github.com/Koenkk/zigbee-herdsman/pull/1556 
 
* https://github.com/Koenkk/zigbee-herdsman/pull/1559

  * https://github.com/Koenkk/zigbee2mqtt/pull/29730

  * https://github.com/Koenkk/zigbee-herdsman-converters/pull/10800

closed 2026-03-16 by Hedda

hm2dev, 2026-10-23:
Is there still someone working on this? I would be willing to help, as I would like to migrate two old Philips Living Colors lamps (LLC001 and LLC007) away from the Hue ecosystem to Home Assistant but those only support touch link for the initial adding flow.

puddly, 2026-10-23:
I don't think anyone is at the moment. Do your devices support the Hue on/off pattern to reset? https://www.reddit.com/r/philipshue/comments/198bzip/solved_how_to_reset_philips_hue_light_without/kibx12j/

TheJulianJES, 2026-10-23:
@hm2dev The older LivingColors (pre-Hue) models with remotes cannot be paired with ZHA at the moment, sorry. TouchLink pairing is not supported.

But I really would not recommend to put them on your main network anyway, as they do not support firmware updates (not even via the Hue Bridge), and the firmware on them is ancient and has a few issues.