zigpy/zigpy #1470: Purpose of currently registered endpoints and server/client clusters

puddly, 2024-09-12:
I'm looking through our current `register_endpoints` code and am trying to document why multiple endpoints exist and why we need each cluster. So far:

```python
    async def register_endpoints(self) -> None:
        """Registers all necessary endpoints.
        The exact order in which this method is called depends on the radio module.
        """

        await self.add_endpoint(
            zdo_types.SimpleDescriptor(
                endpoint=1,
                profile=zigpy.profiles.zha.PROFILE_ID,
                # Is this device type used for something specific?
                device_type=zigpy.profiles.zha.DeviceType.IAS_CONTROL,
                device_version=0b0000,
                # Server clusters
                input_clusters=[
                    # I think this cluster is mandatory to have on at least one endpoint
                    zigpy.zcl.clusters.general.Basic.cluster_id,
                    # For remotes, I think?
                    zigpy.zcl.clusters.general.OnOff.cluster_id,
                    # Required for Develco SPLZB-131
                    zigpy.zcl.clusters.general.Time.cluster_id,
                    # I guess because we support OTA?
                    zigpy.zcl.clusters.general.Ota.cluster_id,
                    # Why this?
                    zigpy.zcl.clusters.security.IasAce.cluster_id,
                ],
                # Client clusters
                output_clusters=[
                    # ???
                    zigpy.zcl.clusters.general.PowerConfiguration.cluster_id,
                    # I believe this is used by many IKEA remotes?
                    zigpy.zcl.clusters.general.PollControl.cluster_id,
                    # Required by Bosch ISW-ZPR1-WP13
                    zigpy.zcl.clusters.security.IasZone.cluster_id,
                    # ???
                    zigpy.zcl.clusters.security.IasWd.cluster_id,
                ],
            )
        )

        await self.add_endpoint(
            zdo_types.SimpleDescriptor(
                endpoint=2,
                profile=zigpy.profiles.zll.PROFILE_ID,
                device_type=zigpy.profiles.zll.DeviceType.CONTROLLER,
                device_version=0b0000,
                input_clusters=[zigpy.zcl.clusters.general.Basic.cluster_id],
                output_clusters=[],
            )
        )
```

dmulcahey, 2024-09-12:
OK here is where IAS came from: https://github.com/zigpy/bellows/pull/135

dmulcahey, 2024-09-12:
So a lot of them came from this commit: https://github.com/zigpy/zigpy/pull/848/commits/8593f1ab36ab43bf696beec0095ad5fb64512680 on this PR: https://github.com/zigpy/zigpy/pull/848

Did we just copy this from zsmart or z2m? we may have to go through their repos to figure out why...

puddly, 2024-09-16:
zigpy-znp started with this one:

```python
    await self._znp.request(
        c.AF.Register.Req(
            Endpoint=1,
            ProfileId=zigpy.profiles.zha.PROFILE_ID,
            DeviceId=zigpy.profiles.zha.DeviceType.IAS_CONTROL,
            DeviceVersion=0b0000,
            LatencyReq=c.af.LatencyReq.NoLatencyReqs,
            InputClusters=[
                clusters.general.Basic.cluster_id,
                clusters.general.Ota.cluster_id,
            ],
            OutputClusters=[
                clusters.security.IasZone.cluster_id,
                clusters.security.IasWd.cluster_id,
            ],
        ),
        RspStatus=t.Status.SUCCESS,
    )

    await self._znp.request(
        c.AF.Register.Req(
            Endpoint=2,
            ProfileId=zigpy.profiles.zll.PROFILE_ID,
            DeviceId=zigpy.profiles.zll.DeviceType.CONTROLLER,
            DeviceVersion=0b0000,
            LatencyReq=c.af.LatencyReq.NoLatencyReqs,
            InputClusters=[clusters.general.Basic.cluster_id],
            OutputClusters=[],
        ),
        RspStatus=t.Status.SUCCESS,
    )
```

How the others ended up in the zigpy PR is very much lost...

closed 2025-03-22 by github-actions[bot]