zigpy/zigpy #1471: AttributeError: 'SizePrefixedSimpleDescriptor' object has no attribute 'is_end_device'

pipiche38, 2024-09-13:
This is happening after pairing a new device ( Tuya Temp+Humi End Device ) with the latest zigpy release

> zigpy==0.66.0
> zigpy_znp==0.12.4

```
2024-09-13 17:24:06,000 WARNING :Device <Device model=None manuf=None nwk=0xDA01 ieee=a4:c1:38:d8:07:d9:a8:e0 is_initialized=False> failed to initialize due to unexpected error
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 205, in initialize
    await self._initialize()
  File "/usr/local/lib/python3.11/site-packages/zigpy/util.py", line 136, in retry
    return await func()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 234, in _initialize
    status, _, endpoints = await self.zdo.Active_EP_req(self.nwk)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 314, in request
    if expect_reply and (self.node_desc is None or self.node_desc.is_end_device):
                                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'SizePrefixedSimpleDescriptor' object has no attribute 'is_end_device'
```

puddly, 2024-09-13:
`node_desc` is a node descriptor, not a simple descriptor. I think you have added code to modify it?

pipiche38, 2024-09-13:
Not sure I follow you.
Yes agree that node_desc is not Simple descriptor, but this looks happening while device initialization.

dmulcahey, 2024-09-13:
The node descriptor is fully managed internally in Zigpy. I think what we’re trying to say is you must have custom code somewhere that is assigning something to `node_desc` on your end… https://github.com/zigpy/zigpy/blob/754b1cde78eb90dd21c11f50b4c9c39bbe936d99/zigpy/device.py#L188 This is all self contained… its either that or you have a device with a firmware that is broken like no other we have ever encountered….

pipiche38, 2024-09-13:
we are not over-writing those methods. 
The device is a Tuya TS0601 Temp_Humi

dmulcahey, 2024-09-13:
OK, can you get a pcap of the traffic? Also, do you also use quirks in addition to Zigpy?

puddly, 2024-09-13:
A full debug log would be nice too

pipiche38, 2024-09-13:
No we don't use quirk. the Zigbee for Domoticz plugin is managing the strange devices in an other maner.

dmulcahey, 2024-09-13:
Can you get a pcap of the init of this device when it is paired? Or a full set of debug logs?

closed 2024-10-14 by puddly