zigpy/zigpy #1476: Hue changed their CA certificates

Glennmen, 2024-09-30:
I came along this PR -> https://github.com/Koenkk/zigbee-OTA/pull/555
It looks like Hue changed their self signed cert and this might have broken the Z2M Remote OTA provider -> https://github.com/zigpy/zigpy/blob/563dcfe22c0bce863808ac97031d4047e77f0905/zigpy/ota/providers.py#L686-L700

Not sure how I should verify this, happy to provide any more information if needed.

puddly, 2024-10-08:
I'm still getting updates for Hue devices. I think they reverted the CA change?

Glennmen, 2024-10-08:
I see updates in HA but so far I haven't been able to install them 🤔 I thought it was related to this CA cert change.
Not sure how I can verify.

puddly, 2024-10-08:
So far it's working fine:

[image]

puddly, 2024-10-08:
> I see updates in HA but so far I haven't been able to install them

Can you enable ZHA debug logging before clicking the "Install" button, then disable it after the update fails to generate a log?

Glennmen, 2024-10-08:
File was too big for Github 😅  I hope this is okay.
https://drive.google.com/file/d/1kszBG34sK4hEA15JqSOhfU2GzDiWfWKm/view

puddly, 2024-10-08:
Looks like Multiprotocol is crashing when OTA starts. Multiprotocol is no longer being developed so if it's breaking for you, migrate back to Zigbee firmware: https://connectzbt1.home-assistant.io/procedures/disable-multiprotocol/

Glennmen, 2024-10-08:
I didn't know it was discontinued 😅  I guess I missed the announced.
Disabling multiprotocol did indeed solve my issue, I will let you/maintainers decide if this ticket is still needed or can be closed.

closed 2024-10-14 by puddly