zigpy/zigpy #1493: sqlite3.OperationalError: near "<": syntax error

stevedh42, 2024-10-28:
Got this error after upgrading zigpy using pip so not sure if it has been fixed since, although the code that caused this issue was added 6 months ago and hasn't neem changed since.

Tracked this issue to

async def load(self) -> None:

appdby.py

in this piece of code.

        await self._load_attributes(
            f"""
                cluster_type={ClusterType.Server}
            AND cluster_id={Basic.cluster_id}
            AND (
                   attr_id={Basic.AttributeDefs.manufacturer.id}
                OR attr_id={Basic.AttributeDefs.model.id}
            )
            """
        )
        
I fixed it by changing   
cluster_type={ClusterType.Server}
to
cluster_type={ClusterType.Server.value}

I don't really program python and don't know the bigger picture so don't know if that is a correct fix.
before that it was producing this in the query 

'SELECT * FROM attributes_cache_v13 WHERE cluster_type=<ClusterType.Server: 0> AND cluster_id=0 AND ( attr_id=4 OR attr_id=5 )

And I assumed  cluster_type= should be using the value of the enum and not the enum itself.

However as I seem to be the only person effected by this 6th month old update it may have somehow been fixed elsewhere and I am just misunderstanding something to do with python of which I am not very familuar.

puddly, 2024-10-29:
How are you running ZHA or zigpy?

stevedh42, 2024-11-01:
Via 'zigbee for domoticz' in domoticz

Hedda, 2024-11-04:
> Via 'zigbee for domoticz' in domoticz

@pipiche38 this should probably be reporting as an issue to zigbeefordomoticz projects Domoticz-Zigbee repository instead, or?

* https://github.com/zigbeefordomoticz/Domoticz-Zigbee

  * https://www.domoticz.com/wiki/ZigbeeForDomoticz

stevedh42, 2024-11-04:
I believe this is the correct place to report it as the error is created entirely within zigpy code.

The cause of the sqlight3 is due to a syntax error in a query created in appdby.py which constructs this query

'SELECT * FROM attributes_cache_v13 WHERE cluster_type=<ClusterType.Server: 0> AND cluster_id=0 AND ( attr_id=4 OR attr_id=5 )

and sqlight3 thinks 
cluster_type=<ClusterType.Server: 0>
is an error.. 

all other examples I have seen elsewhere do
cluster_type=<ClusterType.Server: 0>

but I don't know enough about sqlight3 syntax to know if using a : instead of an = is the cause of the issue..

all I know is hat when I changed the line

cluster_type={ClusterType.Server}

in 
appdby.py

to
cluster_type={ClusterType.Server.value}

the error went away.

but I may just have a bad version of sqlight3 or something....

so 1).

should

'SELECT * FROM attributes_cache_v13 WHERE cluster_type=<ClusterType.Server: 0> AND cluster_id=0 AND ( attr_id=4 OR attr_id=5 )

be a valid sqlight3 query..

as for me only

'SELECT * FROM attributes_cache_v13 WHERE cluster_type=0 AND cluster_id=0 AND ( attr_id=4 OR attr_id=5 )

worked.

puddly, 2024-11-04:
The query is constructed with a Python f-string. In your Python version, the following code apparently fails:

```python
import enum

class ClusterType(enum.IntEnum):
    Server = 0
    Client = 1

assert f"{ClusterType.Server}" == "0"
```

`enum` is built into Python and I'm unable to reproduce this on any version of Python I have access to.

We thoroughly unit and integration test the database code with supported Python release: 3.9, 3.10, 3.11, 3.12, and 3.13. This error would cause *every* zigpy installation to fail to load and with the number of installations we have, it's a little odd that you're the only one to ever report this problem.

stevedh42, 2024-11-04:
Thanks, I guessed it is possibly something in my setup.

if possible, could you help me I understand where mine is going wrong.

on mine

cluster_type={ClusterType.Server}

is being converted to the string,

cluster_type=<ClusterType.Server: 0>

and sqlight3 is complain when it hits the < and throwing a syntax error.

if I change it to cluster_type={ClusterType.Server.value}

that then is converted to 

cluster_type=0

which works fine

I was aware that ClusterType is an enum

but was not sure exactly how they work in python, so

should 
cluster_type={ClusterType.Server} produce cluster_type=<ClusterType.Server: 0> and is my issue with sqlight3

or should 

cluster_type={ClusterType.Server} produce cluster_type=0

and my issue is on the python side of things.

thanks for your time.

stevedh42, 2024-11-05:
Ok.. looks definitely an issue my end.. and possibly due to the environment it is running in.
Sorry for taking you time... If you can give me until the end of the weekend to track down the actual issue. and thanks for the comments as they gave me some clues where to look. I now know the enum should just return 0.

puddly, 2024-11-26:
I'm going to go ahead and close this issue, as so far there hasn't been any other report of this problem.

closed 2024-11-26 by puddly

stevedh42, 2024-11-27:
Just adding a comment, not sure if worth re-opening, but in trying to work out what was happening I notice that pythons handling of converting enums to strings has varied through various versions and that in my opinion '.value' when a value is required is safer as in some environments as without that you can sometimes get the whole enum instead of just the value.

pipiche38, 2025-07-06:
I have one user whom have reported this issue today. Will try to get more details

puddly, 2025-07-06:
@pipiche38 this is likely caused by using an unsupported Python version, not something in the zigpy codebase.

pipiche38, 2025-07-06:
Python Version: 3.12.3  ? not supported ?

Chrominator, 2025-07-07:
Hello,
I do confirm having this issue when running zigpy through Domoticz-Zigbee stable 7.1.022 with python 3.12.3 (default python under Ubuntu 24.04.2)
And thanks to @stevedh42, I did the change [he specified above](https://github.com/zigpy/zigpy/issues/1493#issue-2617653046) and zigpy is running like a charm.