zigpy/zigpy #1503: Manufacturer override on custom cluster does not work ?

papylhomme, 2024-11-18:
Hello,

when trying to debug a quirk for a weird device we tried to use the `manufacturer_id_override` field but it didn't help with forcing the code used when sending commands or reading/writing attributes.

For more details, original discussion is here https://github.com/zigpy/zha-device-handlers/pull/3364#issuecomment-2482896472 : 
- the device is a clone of another one
- it announces a manufacturer_id of `4727`
- manufacturer specific cluster respond only with a manufacturer_code of `4747` (the one from the original device)

I'm new to zigpy / zha, but I had expected the override to take over in this case, or is intended for another purpose ?

puddly, 2024-11-19:
What happens if you set `is_manufacturer_specific=True` on the attribute definition?

papylhomme, 2024-11-20:
> What happens if you set `is_manufacturer_specific=True` on the attribute definition?

The attribute is declared with the flag activated as follow

```python
        pilot_wire_mode = ZCLAttributeDef(
            id=0x0000,
            type=NodOnPilotWireMode,
            zcl_type=DataTypeId.uint8,
            is_manufacturer_specific=True,
        )
```

I played with the zigby code, and from what I gather  the `manufacturer_id_override` is not used where we want. It is [used in the Device class when present](https://github.com/zigpy/zigpy/blob/d0fa17bfcb5bfbda47146ec7dd6ef49dbedc298b/zigpy/device.py#L643), but the CustomCluster class reads [directly from the value provided in the endpoint](https://github.com/zigpy/zigpy/blob/d0fa17bfcb5bfbda47146ec7dd6ef49dbedc298b/zigpy/quirks/__init__.py#L378)

puddly, 2024-11-20:
Ah, I see what you mean. `manufacturer_id_override` is present on `CustomCluster` but it doesn't actually get used, only `Device.manufacturer_id_override` does. This is a bug.

closed 2024-11-25 by puddly