zigpy/zigpy #1508: Custom v2 quirks cause `MultipleQuirksMatchException` with built-in ones

TheJulianJES, 2024-11-25:
## Issue

As seen in https://github.com/zigpy/zha-device-handlers/pull/3541#issuecomment-2497597198, zigpy currently cannot load custom v2 quirks when there's already a built-in v2 quirk for the same device. This isn't an issue with v1 quirks, so we should adjust this for v2 quirks.

This is somewhat critical to have users test new v2 quirks. A workaround for now is to remove the
```python

## Additional information

zha-quirks loads both the normal quirks and the custom quirks afterwards: [zhaquirks/__init__.py#L449-L489](https://github.com/zigpy/zha-device-handlers/blob/e5582c347c48a7b9a903515c0f5b17d184c38557/zhaquirks/__init__.py#L449-L489)
To fix issues when reloading ZHA in HA, we also purge custom quirks then.
We might be able to use similar logic to the custom quirks purging logic implemented in [zigpy/pull/1451](https://github.com/zigpy/zigpy/pull/1451/files#diff-82981555dfaad7a0d8871cba75d0fc78884a4847a0cb730eab114b9e684f6baa) to detect (and flag?) custom v2 quirks, so we can have custom quirks take precedence over built-in ones here: [zigpy/quirks/registry.py#L124-L127](https://github.com/zigpy/zigpy/blob/d0fa17bfcb5bfbda47146ec7dd6ef49dbedc298b/zigpy/quirks/registry.py#L124-L127).

Also thought about if it makes sense to mark all custom quirks (or built-in ones) with a special flag when loading, but we can also detect it later (similar to the custom quirks purging logic).

TheJulianJES, 2024-11-25:
A somewhat related issue (with v1) quirks is https://github.com/zigpy/zha/issues/299. Apparently device automation triggers from built-in quirks take precedence over custom quirks in HA. This the wrong way around.

TheJulianJES, 2024-12-09:
According to https://github.com/zigpy/zha-device-handlers/issues/3589#issuecomment-2528253103, this also happens when there's just a custom v2 quirk and ZHA is then reloaded. I guess we're not purging custom v2 quirks correctly in that case?

erkr, 2024-12-10:
> According to https://github.com/zigpy/zha-device-handlers/issues/3589#issuecomment-2528253103, this also happens when there's just a custom v2 quirk and ZHA is then reloaded.

I think it is relevant that I was evaluating the custom quirk and therefore making changes between reloads

puddly, 2025-01-17:
Can we get rid of `MultipleQuirksMatchException`?

TheJulianJES, 2025-01-24:
This should be fixed with https://github.com/zigpy/zigpy/pull/1534, included in zigpy 0.75.0 and later. Should be part of the Home Assistant 2025.2.0 release.

closed 2025-01-24 by TheJulianJES