zigpy/zigpy #1516: Support for Ikea Styrbar remote control color temperature adjustment

DataBeaver, 2024-12-03:
I have an Ikea Styrbar model E2002.  It has buttons for increasing and decreasing brightness as well as "left" and "right" buttons which adjust the color temperature of the lights when connected directly (or possibly select from a wider variety of colors with RGB lights).  However these buttons provide no detectable cluster events with zigpy.  I get this in the log when pressing one of the buttons:
```
DEBUG:bellows.ash:Received data 7232b1a90d2a7d5e694ee17d31db5581ffed63af263937350e7e
DEBUG:bellows.ash:Received frame DataFrame(frm_num=7, re_tx=0, ack_num=2, ezsp_frame=b'p\x90\x01Y\x00k\xdb\x17u[\xfe\xff\xd4m\xa4\xff\xe1\x01\x92\xda')
DEBUG:bellows.ash:Sending frame AckFrame(res=0, ncp_ready=0, ack_num=0) + FLAG
DEBUG:bellows.ash:Sending data  8070787e
DEBUG:bellows.ezsp.protocol:Received command incomingRouteRecordHandler: {'source': 0xDB6B, 'sourceEui': a4:6d:d4:ff:fe:5b:75:17, 'lastHopLqi': 255, 'lastHopRssi': -31, 'relayList': [0xDA92]}
DEBUG:bellows.zigbee.application:Received incomingRouteRecordHandler frame with [0xDB6B, a4:6d:d4:ff:fe:5b:75:17, 255, -31, [0xDA92]]
DEBUG:bellows.zigbee.application:Processing route record request: (0xDB6B, a4:6d:d4:ff:fe:5b:75:17, 255, -31, [0xDA92])
DEBUG:aiosqlite:executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f8531c77d30>, 'INSERT INTO relays_v13 VALUES (:ieee, :relays)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays WHERE relays != :relays', {'ieee': a4:6d:d4:ff:fe:5b:75:17, 'relays': b'\x01\x92\xda'})
DEBUG:aiosqlite:operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f8531c77d30>, 'INSERT INTO relays_v13 VALUES (:ieee, :relays)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays WHERE relays != :relays', {'ieee': a4:6d:d4:ff:fe:5b:75:17, 'relays': b'\x01\x92\xda'}) completed
DEBUG:aiosqlite:executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f8531c77d30>)
DEBUG:aiosqlite:operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f8531c77d30>) completed
DEBUG:bellows.ash:Received data 0232b1a97d312a15b658914a24ab5593499cb8d84886159874f4c31f98bd793fa6e6cdda44607e
DEBUG:bellows.ash:Received frame DataFrame(frm_num=0, re_tx=0, ack_num=2, ezsp_frame=b'p\x90\x01E\x00\x00\x04\x01\x05\x00\x01\x01\x00\x01\x00\x00\xf6\xff\xe3k\xdb\xff\xff\t\x05|\x11A\x07\x00\x01\r\x00\x04')
DEBUG:bellows.ash:Sending frame AckFrame(res=0, ncp_ready=0, ack_num=1) + FLAG
DEBUG:bellows.ash:Sending data  8160597e
DEBUG:bellows.ezsp.protocol:Received command incomingMessageHandler: {'type': <EmberIncomingMessageType.INCOMING_UNICAST: 0>, 'apsFrame': EmberApsFrame(profileId=260, clusterId=5, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=246), 'lastHopLqi': 255, 'lastHopRssi': -29, 'sender': 0xDB6B, 'bindingIndex': 255, 'addressIndex': 255, 'messageContents': b'\x05|\x11A\x07\x00\x01\r\x00'}
DEBUG:bellows.ezsp.protocol:Frame contains trailing data: b'\x04'
DEBUG:bellows.zigbee.application:Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=5, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=246), 255, -29, 0xDB6B, 255, 255, b'\x05|\x11A\x07\x00\x01\r\x00']
DEBUG:zigpy.application:Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 12, 3, 17, 13, 1, 496649, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDB6B), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=246, profile_id=260, cluster_id=5, data=Serialized[b'\x05|\x11A\x07\x00\x01\r\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-29)
DEBUG:zigpy.device:[0xdb6b] Ignoring message on unknown cluster 5 for endpoint <Endpoint id=1 in=[basic:0x0000, power:0x0001, identify:0x0003, poll_control:0x0020, lightlink:0x1000, manufacturer_specific:0xFC57] out=[identify:0x0003, on_off:0x0006, level:0x0008, ota:0x0019, lightlink:0x1000] status=<Status.ZDO_INIT: 1>>
DEBUG:aiosqlite:executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f8531c77d30>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1733245981.496649, 'ieee': a4:6d:d4:ff:fe:5b:75:17, 'min_update_delta': 30.0})
DEBUG:aiosqlite:operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f8531c77d30>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1733245981.496649, 'ieee': a4:6d:d4:ff:fe:5b:75:17, 'min_update_delta': 30.0}) completed
DEBUG:aiosqlite:executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f8531c77d30>)
DEBUG:aiosqlite:operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f8531c77d30>) completed
```
The important part seems to be the `Ignoring message on unknown cluster 5 for endpoint <Endpoint id=1 in=[basic:0x0000, power:0x0001, identify:0x0003, poll_control:0x0020, lightlink:0x1000, manufacturer_specific:0xFC57] out=[identify:0x0003, on_off:0x0006, level:0x0008, ota:0x0019, lightlink:0x1000] status=<Status.ZDO_INIT: 1>>`.  The endpoint does not advertise a cluster with ID 5, yet the device is sending a message for it.  Cluster 5 seems to be "scenes" - at a quick reading I would guess it's used to store predefined settings and quickly recall them.  This makes sense since the lights have three predefined color temperature settings these buttons cycle through.

The endpoint not being advertised is slightly concerning and may indicate that the device is not functioning according to the specification.  Let me know if you want me to gather any additional information (along with instructions for how to get said information).

MattWestb, 2024-12-03:
If the remote is working OK in the mesh network its sending custom command for changing "scenes" upo or down of the stored scenes in the light like this:
```
event_type: zha_event
data:
  device_ieee: 84:2e:14:ff:fe:5f:89:0e
  unique_id: 84:2e:14:ff:fe:5f:89:0e:1:0x0005
  device_id: 2cd87a51a13b453b07da825de6b453e0
  endpoint_id: 1
  cluster_id: 5
  command: press
  args:
    - 256
    - 13
    - 0
  params:
    param1: 256
    param2: 13
    param3: 0
origin: LOCAL
time_fired: "2024-12-03T18:08:31.534897+00:00"
context:
  id: 01JE6VB8SE082TE378RX8F2JT3
  parent_id: null
  user_id: null
```
with args 256 for right and 257 for left press.
You log is only shoeing router handshake and not real traffic from the device.
Try do one reconfigure from the device card and waking up the device then sending to it by pressing one button.
If not working (getting all green) restart HA and try agen and some time its needed taking out the battery and putting it back for it start reviving commands (its one sleeping end device so it shall sleeping nearly all the time.)
Then the device is working OK you is getting checking in the log every 55 minutes and if not you need getting it configured OK.
```
IKEA Remote Control N2 White Checkin event was fired 
```
The scene switching is only working then binding the controller to lights and they need supporting the custom commands that is IKEA is using so no HUE or tuya.
If have IKEA CWS light they is changing color and WS only color temperature that if pre stored in the firmware (old firmware is loosing then then the device is being retested).
Its possible storing new / more scene in the device but its little trickey.

DataBeaver, 2024-12-03:
The log does contain some router stuff but it seems to me that the latter half is the actual command.  The on and off buttons are giving very similar output.  Here's an example from the on button:
```
DEBUG:bellows.ezsp.protocol:Received command incomingRouteRecordHandler: {'source': 0xDB6B, 'sourceEui': a4:6d:d4:ff:fe:5b:75:17, 'lastHopLqi': 255, 'lastHopRssi': -34, 'relayList': [0xDA92]}
DEBUG:bellows.zigbee.application:Received incomingRouteRecordHandler frame with [0xDB6B, a4:6d:d4:ff:fe:5b:75:17, 255, -34, [0xDA92]]
DEBUG:bellows.zigbee.application:Processing route record request: (0xDB6B, a4:6d:d4:ff:fe:5b:75:17, 255, -34, [0xDA92])
DEBUG:aiosqlite:executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f638d803c40>, 'INSERT INTO relays_v13 VALUES (:ieee, :relays)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays WHERE relays != :relays', {'ieee': a4:6d:d4:ff:fe:5b:75:17, 'relays': b'\x01\x92\xda'})
DEBUG:aiosqlite:operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f638d803c40>, 'INSERT INTO relays_v13 VALUES (:ieee, :relays)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays WHERE relays != :relays', {'ieee': a4:6d:d4:ff:fe:5b:75:17, 'relays': b'\x01\x92\xda'}) completed
DEBUG:aiosqlite:executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f638d803c40>)
DEBUG:aiosqlite:operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f638d803c40>) completed
DEBUG:bellows.ash:Received data 0232b1a97d312a15b658924a24ab5593499c5fd87586159874fec73f88f879937e
DEBUG:bellows.ash:Received frame DataFrame(frm_num=0, re_tx=0, ack_num=2, ezsp_frame=b'p\x90\x01E\x00\x00\x04\x01\x06\x00\x01\x01\x00\x01\x00\x00\x11\xff\xdek\xdb\xff\xff\x03\x01\\\x01\x04')
DEBUG:bellows.ash:Sending frame AckFrame(res=0, ncp_ready=0, ack_num=1) + FLAG
DEBUG:bellows.ash:Sending data  8160597e
DEBUG:bellows.ezsp.protocol:Received command incomingMessageHandler: {'type': <EmberIncomingMessageType.INCOMING_UNICAST: 0>, 'apsFrame': EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=17), 'lastHopLqi': 255, 'lastHopRssi': -34, 'sender': 0xDB6B, 'bindingIndex': 255, 'addressIndex': 255, 'messageContents': b'\x01\\\x01'}
DEBUG:bellows.ezsp.protocol:Frame contains trailing data: b'\x04'
DEBUG:bellows.zigbee.application:Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=17), 255, -34, 0xDB6B, 255, 255, b'\x01\\\x01']
DEBUG:zigpy.application:Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 12, 3, 19, 29, 19, 389214, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDB6B), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=17, profile_id=260, cluster_id=6, data=Serialized[b'\x01\\\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-34)
DEBUG:zigpy.zcl:[0xDB6B:1:0x0006] Received ZCL frame: b'\x01\\\x01'
DEBUG:zigpy.zcl:[0xDB6B:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x01>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=92, command_id=1, *direction=<Direction.Client_to_Server: 0>)
DEBUG:zigpy.zcl:[0xDB6B:1:0x0006] Decoded ZCL frame: OnOff:on()
DEBUG:zigpy.zcl:[0xDB6B:1:0x0006] Received command 0x01 (TSN 92): on()
DEBUG:zigpy.zcl:[0xDB6B:1:0x0006] No explicit handler for cluster command 0x01: on()
DEBUG:aiosqlite:executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f638d803c40>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1733254159.389214, 'ieee': a4:6d:d4:ff:fe:5b:75:17, 'min_update_delta': 30.0})
DEBUG:aiosqlite:operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f638d803c40>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1733254159.389214, 'ieee': a4:6d:d4:ff:fe:5b:75:17, 'min_update_delta': 30.0}) completed
DEBUG:aiosqlite:executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f638d803c40>)
DEBUG:aiosqlite:operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f638d803c40>) completed
```
The difference seems to be that in this case the command is recognized as belonging to an advertised cluster so it's processed further.

I'm not using HomeAssistant, I'm writing code directly on top of zigpy.  The remote control is working, I'm receiving on/off and level control events from it and sending corresponding controls to the lights which react as expected.

[... cut ...]
While the lights advertise a scenes cluster, there seems to be no scenes defined (the count attribute returns 0).  I don't know what data gets sent between the remote control and the lights when paired directly.  `lighting.Color` seems to be the right one to control the color temperature.

So, is this a case of Ikea doing something off-spec and I have no choice but to deal with the decoding the raw binary data?  Or is there something that could be improved in zigpy to get the left and right buttons on the remote control to be recognized?  My lack of knowledge of the zigbee protocol does not allow me to verify this at the moment.

MattWestb, 2024-12-03:
Then the controller is working OK in the mesh network.
IKEA custom things is implanted in zigpy here for the scene cluster:
https://github.com/zigpy/zha-device-handlers/blob/97413e3ec2e5fdbcd17239b810d573cec4d3964d/zhaquirks/ikea/__init__.py#L31-L60
Its only some parameters that is used in the frame you need to catch.

MattWestb, 2024-12-03:
By the way i think the IKEA "default" scenes stored in the firmware is not being reported if you reading the attribute.
If you is binding the scene cluster to one light you can see how its reacting on the commands and is also getting the commands to the system if you have configuring reporting OK of the controller.

closed 2025-06-08 by github-actions[bot]