zigpy/zigpy #1523: Sensor fallback_name doesn't work for SensorDeviceClass.TEMPERATURE

tomasmcguinness, 2024-12-27:
I'm building my own Zigbee sensor device with two temperature clusters (in different endpoints). I'm trying to develop a Quirk that will name then as "Flow Temperature" and "Return Temperature"

I will admit that I'm struggling to understand the QuirkBuilder, so my issues may well stem from ignorance :)

This is what I've got so far

```
(
    QuirkBuilder("coldbear", "FART Sensor")
    .replaces(TemperatureMeasurement, endpoint_id=5)
    .sensor(
        TemperatureMeasurement.AttributeDefs.measured_value.name,
        TemperatureMeasurement.cluster_id,
        endpoint_id=5,
        multiplier=0.01,
        device_class=SensorDeviceClass.TEMPERATURE,
        state_class=SensorStateClass.MEASUREMENT,
        unit=UnitOfTemperature.CELSIUS,
        translation_key="flow_temperature",
        fallback_name="Flow Temperature",
    )
    .replaces(TemperatureMeasurement, endpoint_id=7)
    .sensor(
        TemperatureMeasurement.AttributeDefs.measured_value.name,
        TemperatureMeasurement.cluster_id,
        endpoint_id=7,
        multiplier=0.01,
        device_class=SensorDeviceClass.TEMPERATURE,
        state_class=SensorStateClass.MEASUREMENT,
        unit=UnitOfTemperature.CELSIUS,
        translation_key="return_temperature",
        fallback_name="Return Temperature",
    )
    .add_to_registry()
)
```

The Quirk is being loaded by Home Assistant and applied correctly, but the outcome isn't what I desire. Instead of a device with two sensor entities, I end up with four sensor entities.

[image]

This is my device signature

```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4660,
    "maximum_buffer_size": 108,
    "maximum_incoming_transfer_size": 1617,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 1617,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "5": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0402"
      ],
      "output_clusters": [
        "0x0003"
      ]
    },
    "7": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0402"
      ],
      "output_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "coldbear",
  "model": "FART Sensor",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}
```

Can you please tell me how:
A) I can replace the existing sensors, so I only have two entities.
B) I can name the entities correctly.

TheJulianJES, 2024-12-28:
I think this is an issue in the ZHA integration in HA we can't really fix (easily). The following priority of names occurs:
1. Translation key (handled by HA)
2. Device class name (handled by HA)
3. Fallback name (handled by ZHA integration)

When a quirk is officially added, zha-quirks is bumped in the ZHA lib, and the ZHA lib bump eventually makes it way to HA.
We automatically update `strings.json` in HA with the new translation keys and the `fallback_name` for (US) English.
So, this is only really an issue with custom quirks, as translation keys/values (or the device class name) will always exist upstream.

Normally, HA fully deals with the name (as entities are generally added via HA only). ZHA is a bit special, as there a lot of devices (Tuya) and even custom devices, so it makes sense to provide some (basic) entity support/metadata from the quirks lib. But to even allow the entities to have a name during development at all, we had to introduce `fallback_name` and it'll only fall back to that if HA wasn't able to find a name before (translation key value or device class name).
We can't really change HA without re-implementing what HA internals normally deal with.

tomasmcguinness, 2024-12-28:
Thanks. That kinda makes sense. When I didn't set the device class, the name worked. 

So what I want to achieve isn't possible as it stands with a custom quirk?

Are there any work arounds?

TheJulianJES, 2025-11-26:
This should be fixed in 2025.12.0b0 and later with:
- https://github.com/home-assistant/core/pull/156751

closed 2025-11-26 by TheJulianJES