zigpy/zigpy #1525: No devices in backups

indi81, 2025-01-05:
Hi, I've recently updated my SkyConnect firmware to 7.4.4 to use Ember driver in Z2M/Home Assistant. I thought this would improve on the ezsp coordinator_backup.json but there are still no devices showing up.

Any clue why? My guess is since this is using ezsp and not ember? Seems to be using zigpy in z2m backups as well.

Runnig zigpy ezsp /dev/ttyUSB0 backup gives just one device,
```
    "devices": [
        {
            "ieee_address": "00158d000883733c",
            "nwk_address": "8a62",
            "is_child": true
        }
    ]
}
```

Checking the z2m backup file gives,

```
  },
  "devices": []
}
```

And hardware info,

```
DEVLINKS: >-
  /dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_xxx-if00-port0
  /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.1:1.0-port0
  /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usbv2-0:1.1:1.0-port0
DEVNAME: /dev/ttyUSB0
DEVPATH: >-
  /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS: usb
ID_MODEL: SkyConnect_v1.0
ID_MODEL_ENC: SkyConnect\x20v1.0
ID_MODEL_ID: ea60
ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.1:1.0
ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0-usb-0_1_1_1_0
ID_PATH_WITH_USB_REVISION: platform-fd500000.pcie-pci-0000:01:00.0-usbv2-0:1.1:1.0
ID_REVISION: '0100'
ID_SERIAL: Nabu_Casa_SkyConnect_v1.0_xxx
ID_SERIAL_SHORT: xxx
ID_TYPE: generic
ID_USB_DRIVER: cp210x
ID_USB_INTERFACES: ':ff0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: SkyConnect_v1.0
ID_USB_MODEL_ENC: SkyConnect\x20v1.0
ID_USB_MODEL_ID: ea60
ID_USB_REVISION: '0100'
ID_USB_SERIAL: Nabu_Casa_SkyConnect_v1.0_xxx
ID_USB_SERIAL_SHORT: xxx
ID_USB_TYPE: generic
ID_USB_VENDOR: Nabu_Casa
ID_USB_VENDOR_ENC: Nabu\x20Casa
ID_USB_VENDOR_ID: 10c4
ID_VENDOR: Nabu_Casa
ID_VENDOR_ENC: Nabu\x20Casa
ID_VENDOR_ID: 10c4
MAJOR: '188'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '3079089'
```

MattWestb, 2025-01-05:
EZSP is in ZHA using hacked link key and is not storing any TC-Link key for devices => possible having 64K secure devices in one network (compared to TI chops that is having under hundred secured devices).
I dont knowing how Z2M is doing in there versions but if not using hashed link keys all keys must being stored in the NCP and its limiting how many secured devices you can have in the network and its depends of the hardware (flash / RAM on the chip) and the config of the firmware.
Also it can being Z2M new implementation working in no security mode (first was secure) then its very bad but i dont knowing how they have doing it in the last versions.

indi81, 2025-01-05:
I'm using zstack (and Zbdongle) instead of ezsp  (and Skyconnect) on my primary HA installation and I get a complete device list and link_keys on almost all devices in the coordinator_backup using Z2M (which seems to be using zigpy). I was expecting the same when switching from EZSP driver to Ember in Z2M but the coordinator backup tool in Z2M seems to be using zigpy and ezsp as well, no support for ember. 

I (and z2m as well) probably need to switch to ember-zli or ask for ember support in zigpy.

puddly, 2025-01-05:
Ember and EZSP are the same thing. Z2M just created a distinction because their driver to communicate with EZSP radios only supports newer firmwares.

It's totally normal not to see any listed devices in the backup. The coordinator doesn't need to know about every device on the network and what you see in the backup is all the coordinator is aware of.

closed 2025-01-17 by puddly