zigpy/zigpy #1526: Multistate(Input,Output,Basic) non-compliant with standard

darrencocco, 2025-01-08:
The values
- present_value
- relinquish_default
Are supposed to be uin16 but the code currently has them typed to a float (I think).

See: https://github.com/zigpy/zigpy/blob/07e0451ce5317185e04b38e43aeb9fb168c62f28/zigpy/zcl/clusters/general.py#L1754

See ZCL Spec revision 6, table 3-80 on page 3-102 for the confirmation

puddly, 2025-01-08:
Could you make a PR to fix this?

MattWestb, 2025-01-08:
@darrencocco Download the latest ZCL (Zigbee Cluster Library Specification R8) or the newer Matter one (Matter 1.4 Device Library-Specification, but the changes is coming in ZCL R9) from https://csa-iot.org/developer-resource/specifications-download-request/ so not making changes that is old and not up to date with latest ZCL.

darrencocco, 2025-01-09:
> Could you make a PR to fix this?

Sorry, I barely know Python and don't have the time to learn it right now.

darrencocco, 2025-01-09:
> @darrencocco Download the latest ZCL (Zigbee Cluster Library Specification R8) or the newer Matter one (Matter 1.4 Device Library-Specification, but the changes is coming in ZCL R9) from [...]

ZCL R8 Table 3-85, page 3-118 it still has the same type specification, uint16 for both.

darrencocco, 2025-01-09:
Also Multistate Value doesn't exist in the Matter language, they've replaced it with "Mode Select" I believe.

PR #1588 (2025-04-27): Fix bad multistate attribute data types
Fixes #1526.

We were using `Single` for attributes whose values should be 16-bit unsigned integers. I've also removed incorrectly-present attributes in the `BinaryOutput` cluster.
closed 2025-04-27 by puddly