zigpy/zigpy #1527: How do I change the units/decimal place of a sensor value?

nimbling, 2025-01-09:
My TS0601 sensor outputs its "formaldehyde" values about 100 times too large.

[image]
(I'd be dead if these formaldehyde readings were true)

I'd like to modify the value in here (the quirk it's using) but I do not know how to.

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/air/ts0601_smart_air.py
Reading into this, I think it refers to:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/air/__init__.py

On line 106, I'm guessing I'd like to change the formaldehyde handler to:

```
        22: DPToAttributeMapping(
            TuyaAirQualityFormaldehyde.ep_attribute,
            "measured_value",
            lambda x: x * 1e-8,  # Adjust scaling factor for formaldehyde
        ),
```
(ChatGPT helped me ;) )

But I'm not sure how to test this locally, before I submit a pull request.

Thank you for reading this far!

nimbling, 2025-01-09:
*cough* *cough* https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md
Reading! brb...

nimbling, 2025-01-09:
So, I read that, and I could not understand from it how to test this locally, my apologies.

prairiesnpr, 2025-01-09:
You probably want https://github.com/zigpy/zha-device-handlers/

I actually have a PR to rework these here, https://github.com/zigpy/zha-device-handlers/pull/3651

As far as testing, you need to set up custom quirks. See https://www.home-assistant.io/integrations/zha/, there are also a few guides on the community forums that might help.

nimbling, 2025-01-10:
Thanks! that managed to have it pick up the custom quirks!

I don't think I have restarted my HASS this often in the span of a day before.

I'm editing the "class" definition in the __init__.py right now, but I'm wrestling either really large numbers, or "." periods where I'd want comma's. I feel like I'm close, but the Dashboard still just shows an INT, while expected/reasonable formatting would be "1,23ppm"

ChatGPT can only get me so far :D

I'll just wait for your PR to make it into main, here's hoping it fixes my device too!
THANK YOU! :)

prairiesnpr, 2025-01-10:
I didn't address this, but looks like we will need to.

Looking at the z2m convertor, https://github.com/Koenkk/zigbee-herdsman-converters/blob/40e72bc2515deb67ef35dddab04437342b7b3eb9/src/devices/tuya.ts#L1300. 

I think your issue is that the device reports in `µg/m3`, but ZHA is expecting a value in `ppm`. So we need to do the conversion, `concentration (ppm) = 24.45 x concentration (mg/m3) ÷ 30.026 g/mo`, see https://teesing.com/en/tools/ppm-mg3-converter.

nimbling, 2025-01-10:
Looking at the z2m convertor, it does not seem to convert it at all?

Both my (identical) sensors give a value near 356,
given your formula, that's:
24,45*356/30,026 = 289,88876307 μg/m3
I think that's still a value that's not right?

I checked, and in the EU 1μg/m3 is the recommended maximum value indoors. I can't imagine being at near 300X that and still alive? :D

I've been banging my head against this for two days now :D I love the puzzle. But I'm stuck.

```
class TuyaAirQualityFormaldehyde(FormaldehydeConcentration, TuyaLocalCluster):
    """Tuya Formaldehyde concentration measurement."""
    def update_attribute(self, attr_name: str, value: Any) -> None:
        """Format the formaldehyde reading."""
        if attr_name == "measured_value":
            scaled_value = 24.45 * (float(value)/100) / 30.026
            super().update_attribute(attr_name, scaled_value)
```

This gives me 2.947.745 ppm - while I'm hoping for 2,95 in this case :(

prairiesnpr, 2025-01-10:
Don't forget, ZHA has a fixed multiplier of 1e6 you need to account for, see https://github.com/zigpy/zha/blob/fd6cf2f570f1778a9c0933a56ebf476caafe859d/zha/application/platforms/sensor/__init__.py#L1294

So your value multiplied by 1e-6 gives you 2.95 ppm

nimbling, 2025-01-10:
Thanks! Oh! But... well! I... I'm lost.

So the current *unmodified* implementation of this quirk (with the built in 1e6 multiplier) gives me "363.000.000" - while I was hoping to just multiply that again by 0.001 or something and end up at 3,56 - but I'm failing to get that to work, because it only expects an INT, or something, so it refuses to be a float anywhere in this process.

prairiesnpr, 2025-01-10:
@nimbling Let's test using this one https://github.com/prairiesnpr/zha-device-handlers/blob/tuya-air-test/zhaquirks/tuya/tuya_air_custom.py

nimbling, 2025-01-10:
Thank you so much, should I replace my "__init__.py" with this one or my "ts0601_smart_air.py", or both?

Do I need to install "zigpy.quirks.v2" or the "zhaquirks.tuya.builder" for this to work?

I hope I did not mislead you into thinking I know what I'm doing, I got to this point using trial, lots of error end some grit rather than knowledge 🙏

prairiesnpr, 2025-01-10:
Just place this in your custom quirks directory, remove any other custom quirks you may have for this device, if it gives you an error, about multiple quirks matches found, then add the following after the existing imports.

```python
from zigpy.quirks import DEVICE_REGISTRY

del DEVICE_REGISTRY._registry_v2[("_TZE204_dwcarsat", "TS0601")]
```

If you get some other type of error, post logs, I don't have any of these, so I can't test it.

If you have made changes to the source itself, outside of a custom quirk, revert those.

nimbling, 2025-01-10:
Thanks!
Uploaded it, it logged this:

```
Logger: zhaquirks
Source: /usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py:481
First occurred: 14:02:17 (1 occurrences)
Last logged: 14:02:17

Unexpected exception importing custom quirk 'tuya_air_custom'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py", line 479, in setup
    spec.loader.exec_module(module)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_zha_quirks/tuya_air_custom.py", line 165, in <module>
    .tuya_dp(
     ^^^^^^^
AttributeError: 'NoneType' object has no attribute 'tuya_dp'
```

prairiesnpr, 2025-01-10:
Updated, missed a return

nimbling, 2025-01-10:
That loads!
Now the PPM is "3" - also in the states section of the developer tools.
Do you feel it's worth the effort to make it show (2?) decimals?

nimbling, 2025-01-10:
That loads! but shows a three digit INT again :D

prairiesnpr, 2025-01-10:
You need to import that.

Add this to the imports
```python
from zigpy.quirks.v2.homeassistant.sensor import SensorStateClass
```

nimbling, 2025-01-10:
That, dear stranger on the internet, works like a charm ✨

Thank you!

closed 2025-01-16 by TheJulianJES