zigpy/zigpy #1539: How could be that the device change its behavior if joined ZHA network??

jafar-atili, 2025-01-23:
Hi,

I have unexplained behavior with my Bticino/Legrand dimmer switch, same coordinator, same Zigbee device, same Mini PC running HA OS with Zigbee2MQTT and ZHA, however there is a difference from the device end depends on zigbee network 

### First difference (not that big deal)

**With ZHA** + open the Zigbee network for join, the device led is **pink**
**With Z2M** + open the Zigbee network for join, the device led is **green**

### Second difference (very big deal)

**When the device joined Z2M** - The physical button on the device are behaving as toggle, doesn't matter where you press on the physical switch, the light will toggle (which is the expected behavior)

However, **if the devices is part of ZHA**, the physical buttons are inverted, the upper part of the switch turn off the light, the bottom part turns on the light.

I found response from the vendor [in this link](https://developer.legrand.com/forums/topic/how-to-enable-toggle-mode-on-nlt-in-zigbee-3-0-mode/) relating to this behavior as following:

> The toggle function is specific to the Legrand gateway. When the product joins a ZigBee 3rd party gateway, the up and down buttons are deterministic. So, unfortunately, this parameter can’t be modified
> 
> Have a good day,

The only difference between Z2M and ZHA is the channel used 15 for Z2M and 25 on ZHA, however I don't think this is the case here.

How can the end device distinguish between ZHA and Z2M? and how can I troubleshoot this weird issue.

pipiche38, 2025-01-23:
It is similar with ZigbeeforDomoticz plugin, and the difference between Pink and Green during pairing time, is that in Pink, you have a Zigbee 3.0 pairing, while when Green you have a Legrand/Netatmo pairing.

I guess that Z2M is simulating as ZigbeeforDomoticz the Legrand/Netatmo Hub environment

jafar-atili, 2025-01-23:
Is it something I can tune with ZHA?

puddly, 2025-01-23:
The device sends commands to the coordinator to determine if it's joined to a Legrand hub or not. You need to figure out what exactly it's sending to do this.

pipiche38, 2025-01-23:
In case of Legrand hub, during the pairing time the hub is sending a broadcast which is part of the Legrand hub behaviour. Not having is possible and makes the device strictly behaving ZB3.0 and not enabling Legrand propitiatory features

puddly, 2025-01-23:
Can you link to where you implement this? We can probably implement this by matching the IEEE prefix or model info.

jafar-atili, 2025-01-23:
Cant we figure it out from the source code of Z2M?

pipiche38, 2025-01-23:
Here is what I do on the Zigbee4Domoticz plugin

Legrand track the Operating Time of their device and teh Gateway as well. And They consider that a Device cannot have an operating time greater than the Hub.
The Attribute used for that is 0x0000/0xf000. So at start, the gateway send a Write Attribute

(1) when the plugin is ready, we send a Write Attribute No Responses
     Cluster: 0x0000
     Attribute: 0xf000
     Data Type: 0x23
     Value: 0x00000000
     Manufacturer Specifi: 0x1021

(2) Then time to time, devices are sending Read Attribute (Command 0x00 on Cluster 0x0000, Attribute 0xf000) to the Gateway and in such case you need to respond the Operating Time (number of Second since the Gateway started - more less when you issued the Write Attribute No Response)

TheJulianJES, 2025-01-23:
Writing an attribute on pairing/reconfiguration can easily be implemented in zha-quirks by overriding `apply_custom_configuration`.
I wanna have a look at what Z2M does for this device, what is the "Zigbee model"?

pipiche38, 2025-01-23:
This is for all legrand devices