zigpy/zigpy #1557: Reasons for OTA failure of some devices

GitforZhangXL, 2025-02-24:
Some older devices, after the OTA download progress reaches 100%, there will be an unresponsive phase that may last up to 5 minutes, and then the device will issue an upgrade end request packet, if the device does not receive the upgrade end response packet sent by the coordinator, it will cause the upgrade to fail, I changed the zigpy.ota.manager file, it will MAX_TIME_WITHOUT_PROGRESS If the device is changed to 300, the device can be upgraded successfully. Is it possible to change this duration in the configuration file

puddly, 2025-02-24:
Can you attach a debug log of your device completing the OTA upgrade? Five minutes is a very long time...

GitforZhangXL, 2025-02-24:
[home-assistant.log](https://github.com/user-attachments/files/18941477/home-assistant.log)

puddly, 2025-02-24:
Is this a device you're building?

closed 2025-02-25 by GitforZhangXL