zigpy/zigpy #1561: [Future request] Add possibility to expose zigbee commands as scripts with quirks v2

uvNikita, 2025-03-02:
I wonder if it would be possible to expose Zigbee commands as scripts for the device.
The script would run `zha.issue_zigbee_cluster_command` action with `ieee` parameter calculated for the selected device.

I imagine the interface would look something like this:
```
QuirkBuilder(...)
....
.script(
  endpoint_id=1,
  cluster_id=513,
  cluster_type="in",
  command_type="client",
  command=241,
  description="Script description",
  mode="queued",
  fields={
    'param1': {'description': 'Field 1 description', required': True},
    'param2': {'description': 'Field 2 description'},
  },
)
```

Then the script would be registered with the following `sequence`:

```
sequence = [
  {
    action = "zha.issue_zigbee_cluster_command";
    data = {
      ieee = "{{ device_attr(entity_id, 'identifiers') | selectattr(0, 'eq', 'zha') | map(attribute=1) | first }}";
      endpoint_id = 1;
      cluster_id = 513;
      cluster_type = "in";
      command_type = "client";
      command = 241;
      params = {
        type = 4;
        param1 = "{{ param1 }}";
        param2 = "{{ param2 }}";
      };
    };
  }
];
```

Please let me know if something like this makes sense.

puddly, 2025-03-02:
Interesting. What are you trying to do with this script?

uvNikita, 2025-03-02:
@puddly thank you for a quick reply!

Have the following command available in my thermostats:

https://github.com/zigpy/zha-device-handlers/blob/b87ac8907765303c915b19d915a1ca5ea9a8bdae/zhaquirks/schneiderelectric/thermostat.py#L226

It allows user to override PI heating demand temporarily by specifying `demand`, `duration` and `type` parameters.

Currently, I have to use a low-level `zha.issue_zigbee_cluster_command` whenever I want to override the demand, but it requires me to specify endpoint_id, cluster_id, ieee etc.

I think, currently, it's possible to expose a button in HA that will send a zigbee command (correct me if I'm wrong), but this only works if the command doesn't have parameters, or they can be defined statically. 

I thought it would be nice to be able to also provide scripts that can accept parameters.

I'm still new to zha/zigpy internals so just let me know if it doesn't make much sense 🙂

puddly, 2025-03-02:
We generally won't be able to create scripts in `zigpy/zha` because ZHA is supposed to be a library that can be used outside of Home Assistant. I think it would be possible, however, to wrap device specific functionality that doesn't map to an entity (yet):

```python
SOME_VOL_SCHEMA = vol.Schema({...})

async def some_action(device: zigpy.device.Device, arg1, arg2) -> None:
    result = await device.endpoint[1].some_cluster_name.command(arg1=arg1, arg2=arg2)

    if not result:
        raise RuntimeError("Some error")

    return True

QuirkBuilder(...)
....
.action(
    id="some_action",
    description="This is an action that does something",
    schema=SOME_VOL_SCHEMA,
    callable=some_action
)
```

And then call it from Home Assistant:

```yaml
action: zha.device_action
target:
  device_id: ...  # Or `entity_id`, or `ieee`, etc.
data:
  arg1: 123
  arg2: "abc"
```

closed 2025-09-05 by github-actions[bot]