zigpy/zigpy #1566: OTA fails for ubisys devices

FlorianBruckner, 2025-03-09:
Koenkk/zigbee-OTA#717 fixed an issue in the firmware catalog for ubisys S1 devices. Ubisys provides two different images with the same version, but mutually exclusive hardware constraints.

Previously, OTA for the device would have been offered in HA, but could not be completed because only the wrong image was in the catalog,. It failed with NO_IMAGE_AVAILABLE due to checking the hardware constraint.

Now, both images are in the catalog:

```{
    "fileName": "10F2-7B02-0000-0001-0192020D-spo-fms.ota1.zigbee",
    "fileVersion": 26345997,
    "fileSize": 256474,
    "url": "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/Ubisys/10F2-7B02-0000-0001-0192020D-spo-fms.ota1.zigbee",
    "imageType": 31490,
    "manufacturerCode": 4338,
    "sha512": "20ff31830a17149b353db48afb9ccd356ac75dec04ea0241ef996918471f4b5bc786916caeae79ba75fffbebd6412f06361032a01fd675e85bb931a5fefd7bf3",
    "otaHeaderString": "ubisys S1\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
    "originalUrl": "http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B02-0000-0001-0192020D-spo-fms.ota1.zigbee",
    "hardwareVersionMax": 1,
    "hardwareVersionMin": 0
  },
  {
    "fileName": "10F2-7B02-0002-0007-0192020D-spo-fms.ota1.zigbee",
    "fileVersion": 26345997,
    "fileSize": 256200,
    "originalUrl": "http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B02-0002-0007-0192020D-spo-fms.ota1.zigbee",
    "url": "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/Ubisys/10F2-7B02-0002-0007-0192020D-spo-fms.ota1.zigbee",
    "imageType": 31490,
    "manufacturerCode": 4338,
    "sha512": "27743f96962964d3670c7cc1cf90356d1e77b1277d4cc97a501b01e0600768fd9335f2a9b94efc7c8e4ae892eeebc7a5a2c99ab4b223d0d957282ac66e11c7ac",
    "otaHeaderString": "ubisys S1\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
    "hardwareVersionMin": 2,
    "hardwareVersionMax": 7
  },
```

The log now shows this:

```
Multiple unique OTA images for version 0192020D with specificity 202 exist. It is not possible to tell which image is correct so all 2 of the colliding images will be ignored.
```

puddly, 2025-03-09:
Can you include your device checking in after zigpy broadcasts the image notification? It will have the constraints the device is sending. Perhaps yours doesn't include a hardware version.

FlorianBruckner, 2025-03-09:
Hi @puddly 

Looks like I need debug logging for this, so I don't have the logs at hand.

What I do know (again, no logs available) is that with only the image for the old hardware version in the catalog, HA did offer me the upgrade. Nevertheless, it then failed with a NO_IMAGE_AVAILABLE because the hardwareVersionMin/Max check detected that the image available does not fit. This was due to the check done in https://github.com/zigpy/zigpy/blob/c54c8876e5b1f3fc1ff8a71ba57c25c8bd17436a/zigpy/ota/__init__.py#L174, which would hint at the device sending its hardware version. I did debug it to that point when I was looking into this issue why my devices was not updating.

I'll try to generate debug logs tomorrow to support that.

TheJulianJES, 2025-03-10:
I believe we only get the min/max hardware version from the OTA file header (where it should be...).
Currently, we ignore the Z2M properties: `hardwareVersionMin`/`hardwareVersionMax`.

closed 2025-03-10 by puddly

FlorianBruckner, 2025-03-11:
Hi @puddly 

thanks for the fix, but I believe there might be an earlier opportunity to catch this case. I added some additional logging, and the problem with duplicate images happens before the OTA is started, while the image is not available.

This is logging in check_compatibility, and this appears to be called for every device/firmware combination when HA tries to identify potential updates - here is the check for the device and image combination:

```
2025-03-11 10:21:55.703 INFO (MainThread) [zigpy.ota] check_compatibility OtaImageWithMetadata(metadata=RemoteOtaImageMetadata(file_version=26345997, manufacturer_id=4338, image_type=31490, checksum='sha512:20ff31830a17149b353db48afb9ccd356ac75dec04ea0241ef996918471f4b5bc786916caeae79ba75fffbebd6412f06361032a01fd675e85bb931a5fefd7bf3', file_size=256474, manufacturer_names=(), model_names=(), changelog=None, release_notes=None, min_hardware_version=None, max_hardware_version=None, min_current_file_version=None, max_current_file_version=None, specificity=None, source='Remote Z2M provider (https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json)', url='https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/Ubisys/10F2-7B02-0000-0001-0192020D-spo-fms.ota1.zigbee', ssl_ctx=<ssl.SSLContext object at 0x7f5c59e3c0>), firmware=None) QueryNextImageCommand(field_control=<FieldControl.HardwareVersion: 1>, manufacturer_code=4338, image_type=31490, current_file_version=17760770, hardware_version=6)
Stack (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once
    handle._run()
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.13/site-packages/zigpy/util.py", line 333, in catching_coro
    return await target
  File "/usr/local/lib/python3.13/site-packages/zigpy/zcl/clusters/general.py", line 2249, in _handle_query_next_image
    images_result = await device.application.ota.get_ota_images(device, cmd)
  File "/usr/local/lib/python3.13/site-packages/zigpy/ota/__init__.py", line 443, in get_ota_images
    if img.check_compatibility(device, query_cmd)
  File "/usr/local/lib/python3.13/site-packages/zigpy/ota/__init__.py", line 156, in check_compatibility
    _LOGGER.info("check_compatibility %s %s", self, query_cmd, stack_info=True)

```

It just logs the image metadata and the device query_cmd, from this you can see that the device does in fact send the hardware_version, but the images metadata don't have the constraints declared in the catalog.

When the update scan fails, the device will show the firmware update as "unavailable":

[image]

And the log (still on the old version without your fix) 

```
Multiple unique OTA images for version 0192020D...
```

@TheJulianJES 

While I agree that the constraints should eventually be checked according to the images header (and as far as I understand, thats the case already when starting OTA for one device), the scan for potential updates has just access to the catalog metadata and not the image.

FlorianBruckner, 2025-03-11:
#1568 adresses the issue with catalog metadata, i have therefore closed my PR.