zigpy/zigpy #1570: Ubisys OTA appears to fail but actually succeeds

FlorianBruckner, 2025-03-11:
Ubisys devices do install the new image after it has been transmitted, but the logs and the UI show error messages:

```
2025-03-11 12:30:14.067 INFO (MainThread) [zigpy.device] [0xdf4b] OTA upgrade progress: (256168 / 256200): 99.9875%
2025-03-11 12:30:19.770 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [546817118496] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/update.py", line 188, in async_install
    await self.entity_data.entity.async_install(version=version)
  File "/usr/local/lib/python3.13/site-packages/zha/application/platforms/update.py", line 301, in async_install
    raise ZHAException(f"Update was not successful: {result}")
zha.exceptions.ZHAException: Update was not successful: <Status.FAILURE: 1>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 489, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/zha/update.py", line 190, in async_install
    raise HomeAssistantError(exc) from exc
homeassistant.exceptions.HomeAssistantError: Update was not successful: <Status.FAILURE: 1>

```

**This really is just a cosmetic issue** and probably the device is misbehaving by not responding to the upgrade call.

Initially, HA offers the previous upgrade but after some minutes, finds the new version on the device and shows the status correctly (in this case, because there is a series of upgrades to be done, the next upgrade until the most recent version is installed).

Hedda, 2025-03-12:
FYI, slightly off-topic but developers working on Ubisys Zigbee OTA might be interested in looking at this related previous request:

* https://github.com/zigpy/zigpy/issues/988

FlorianBruckner, 2025-03-12:
@Hedda 

ubisys firmware images are available in https://github.com/Koenkk/zigbee-OTA, so these images are available with the `z2m` type provider (see https://github.com/zigpy/zigpy/wiki/OTA-Configuration).