zigpy/zigpy #1572: Add device name (or ieee if no device name) to NWK.

mkz212, 2025-03-14:
In logs there is only NWK. Displaying device name will make it easier to diagnose problem.

https://github.com/zigpy/zigpy/blob/7957497f5184d888e18586ee8007d1d2fe04f688/zigpy/endpoint.py#L318-L321

Maybe this will add device name (or ieee if device name not available) before NWK?

```py

def log(self, lvl: int, msg: str, *args: Any, **kwargs: Any) -> None:
    device_name = self._device.name if hasattr(self._device, 'name') and self._device.name else str(self._device.ieee)
    msg = "[%s:0x%04x:%s] " + msg
    args = (device_name, self._device.nwk, self._endpoint_id, *args)
    LOGGER.log(lvl, msg, *args, **kwargs)
```

MattWestb, 2025-03-14:
Its some cases the coordinator / system / the zigbee network is only using only NWK or IEEE.
The last is mostly with network searing commands like one device is rejoining and have change the NWK (i think that is one bug in Silabs SDK that is doing the device think the NWK is in use and trying with one new random one then rejoining) and its not easy finding witch device that is messing around then the NWK in the log is not the actual the system is using now. But its one hierarchy how address types is being used in Zigbee and in the end its better little more formation then trying finding out what is happening in the system.

mkz212, 2025-03-15:
Hmm. Thanks for the answer, although I don't understand too much. I only want the logs to include not only NWK because after that it is difficult to identify which device generates a warning in logs. 

So instead of:
`[0xC34C:1:0x0020] error message`
to have e.g.:
`[Bedroom light:0xC34C:1:0x0020] error message`.

MattWestb, 2025-03-15:
The problem with timeout then checking OTA is normally happening then the sleeping end device is doing one checking (every 50 min if being real Zigbee 3 devices) and waiting the coordinator shall acking it with fast pull time out,
The problem have being around for nearly 2 years now and i thinks it was implanted then changing the com libery. The end device is sending the command to its parent and waiting for getting one ack and under that time its pulling the parent fast. But somthing (the mesh network or the coordinator / host system) is having some lock that is doing the command is not arriving in time and the end device is going back to sleep. And normally the system is trying to pulling the device for OTA at this timethen it shall being "online" but its not that it shall have being if all war working OK.

I have it in my production system (90 + devices normally stable) and my test network (30+ devices all IKEA routers so true Zigbee 3).

And i agree with you its much better see the device name and not need looking around for finding it out witch it is having the problem.

mkz212, 2025-03-15:
> The problem with timeout then checking OTA is normally happening then the sleeping end device is doing one checking (every 50 min if being real Zigbee 3 devices) and waiting the coordinator shall [...]

Thanks for such a thorough explanation. I know that my network needs a few corrections. But it would be much easier for me to determine what to improve if the name of the device was in the logs. I think this is a small change and probably easy to implement, but a change that can have a huge impact on improving the performance of our networks.

puddly, 2025-03-16:
Since you opened two issues, I'm going to close this one and continue the discussion here https://github.com/home-assistant/core/issues/140529

closed 2025-03-16 by puddly