zigpy/zigpy #1573: Zigpy as in HA 2025.3.3: Cannot make zigpy recognize new firmware

galmok, 2025-03-16:
I am trying to update my smart plugs from Develco and I had an old configuration that automatically updated my plugs from my zigpy_ota folder based on a simple configuration in my HA yaml file. Since then, the format has changed and I cannot make HA/zigpy recognize the firmware anymore. It recognizes errors in the configuration (like using from type or missing warning) but otherwise completely ignores my settings.

I have tried to follow the manual and even tried looking at the source code, but so far haven't found the reason for my failure to make it work. 

The failure is showing upon start of HA when I check the firmware of the devices. HA claims there is no new firmware available, but I know there is, because I have one and my previously installed devices all have that firmware installed with the help of the old version of HA.

I am using the container version of HA.

These are my attempts at json files for the .zigbee firmware file:

```
root@raspberrypi:~/docker/homeassistant/config/zigpy_ota# cat version_info.json
[
    {
        "binary_url": "file:///config/zigpy_ota/ZR - Smartplug New - Release-SSIG_3.12.16.zigbee",
        "file_version": 199696,
        "image_type": 736,
        "manufacturer_id": 4117,
        "changelog": "",
        "checksum": "sha3-256:c06f964e0ce748a987fdb49af325f218da526049993a8f5b3795c8b89478b46d"
    }
]
root@raspberrypi:~/docker/homeassistant/config/zigpy_ota# cat index.json
{
  "firmwares": [
    {
      "fileName": "ZR_Smartplug_SSIG_3.12.16.zigbee",
      "fileVersion": 199696,
      "fileSize": 181164,
      "url": "file:///config/zigpy_ota/ZR - Smartplug New - Release-SSIG_3.12.16.zigbee"
      "imageType": 736,
      "manufacturerCode": 4117,
      "sha512": "eaf4e925dbdf32f171fb8323c52b8652620dea5d62bc007e61ee61d276bdd90f097400a1c1bc48980d78c276117415e274c34529b481be82c28cd3695e2be817",
      "otaHeaderString": "\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
    }
  ]
}
root@raspberrypi:~/docker/homeassistant/config/zigpy_ota# cat index2.json
{
  "firmwares": [
    {
      "fileName": "ZR_Smartplug_SSIG_3.12.16.zigbee",
      "fileVersion": 199696,
      "fileSize": 181164,
      "path": "file:///config/zigpy_ota/ZR - Smartplug New - Release-SSIG_3.12.16.zigbee"
      "imageType": 736,
      "manufacturer_id": 4117,
      "checksum": "sha3-256:c06f964e0ce748a987fdb49af325f218da526049993a8f5b3795c8b89478b46d"
    }
  ]
}
```

And this is the combined effort in my configuration.yaml:

```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
  enable_quirks: true
  database_path: /config/zigbee.db
  zigpy_config:
    ota:
      extra_providers:
        - type: zigpy_local
          index_file: /config/zigpy_ota/version_info.json
        - type: zigpy_local
          index_file: /config/zigpy_ota/index2.json
        - type: zigpy_remote
          url: "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/refs/heads/master/index.json"
          manufacturer_ids: [ 0x1015 ]
        - type: advanced
          warning: I understand I can *destroy* my devices by enabling OTA updates from
            files. Some OTA updates can be mistakenly applied to the wrong
            device, breaking it. I am consciously using this at my own risk.
          path: /config/zigpy_ota
```

Nothing has worked and I suspect there has to be a bug in the system as I cannot guess at more things to try than reaching out here.

puddly, 2025-03-16:
If these are hosted in the Z2M repo, just add the `- type: z2m` provider and zigpy will use the repo directly.

puddly, 2025-03-16:
Just to confirm, your config should look like this:

```yaml
zha:
  custom_quirks_path: /config/custom_zha_quirks/  # Remove if you have none
  zigpy_config:
    ota:
      extra_providers:
        - type: z2m
```

When you check for updates, you click the refresh button at the top right of the update page (http://homeassistant.local:8123/config/updates)?

galmok, 2025-03-16:
I haven't seeing that link before but when I visited it, I was greeted with an update message for one of the devices that needed an upgrade. I then checked config/dashboard and then the update also was visible there. I know I have seen this page many time but never has the update been presented to me. First when I went to config/updates did I get the update presented.

But I am happy that at least one of my devices is getting updated now. I hope the other one also is offered soon.

And I am still confused about the quirk on one of my plugs, but I'll open another case for that if I cannot find something in my files about this.

galmok, 2025-03-16:
And the other one was also detected after a refresh. 

To be honest, I was expecting the firmware section on the specific device page to report if a new firmware was available and it never did so. The refresh apparently is required for this to be checked. I can only wonder now how many of my attempts actually were good, but just missed the Refresh...

puddly, 2025-03-16:
OTA updates for Zigbee are very chatty and it's basically a network-stalling operation to even check for updates. Zigpy instead checks small portions of your devices for OTA updates in the background, covering your whole network in about 8 hours (unless it is very small).

Normally, you'll be notified about updates over time. You can speed it up by clicking the "check for updates" button in the Home Assistant update page, which sends a network-wide broadcast and asks every device to check in.

galmok, 2025-03-16:
Maybe HA/Ziggy could report the latest available firmware correctly on the device usage? This wouldn't require interrogating the devices and could provide the user with the knowledge that the information in configuration.yaml was entered correctly? I used many hours trying to get this configuration right...

puddly, 2025-03-16:
If click on the update entity, it will show you the currently installed and latest available versions.

[image]

galmok, 2025-03-17:
That is what I used to check if I had it configured correctly, but it never showed me that the updated firmware was available. Combined with no error in the log, I was stumped.

But I never clicked in the refresh button and made fast iterations (every 2-3 minutes sometimes). And no documentation said I had to do so or wait up to 8 hours per attempt. 

Is it possible to make the firmware update check in just 1 device?

puddly, 2025-03-18:
> Is it possible to make the firmware update check in just 1 device?

Not at the moment, Home Assistant update entities are all or nothing right now. If Home Assistant adds a way to do this in the future, we can do that.

closed 2025-03-20 by puddly