zigpy/zigpy #1647: WARNING (MainThread) [zigpy.device]

Dlivera, 2025-08-11:
2025-08-11 15:01:18.861 WARNING (MainThread) [zigpy.device] Cluster 0x000a on <Device model='TS0207' manuf='_TZ3000_nkkl7uzv' nwk=0x370D ieee=88:1a:14:ff:fe:02:fd:c3 is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster). Please report this here: https://github.com/zigpy/zigpy/issues/1640

Hedda, 2025-09-18:
zigpy/zha devs collect incorrect direction warning messages in a separate issue so you should post there instead and then close this:
 
- https://github.com/zigpy/zigpy/issues/1640

TheJulianJES, 2025-10-06:
Thanks for the report! Device is already noted in https://github.com/zigpy/zigpy/issues/1640.
For now, please ignore the message.