zigpy/zigpy #1649: Cluster ... has incorrect direction warning

Private-Storm, 2025-08-12:
Cluster 0x000a on <AGL001 model='lumi.airrtc.agl001' manuf='LUMI' nwk=0xD205 ieee=54:ef:44:10:00:b8:36:13 is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster). Please report this here: https://github.com/zigpy/zigpy/issues/1640

Hedda, 2025-09-08:
zigpy/zha devs collect incorrect direction warning messages in a separate issue so you should post there instead and then close this:
 
- https://github.com/zigpy/zigpy/issues/1640

TheJulianJES, 2025-10-06:
Thanks for the report! Device is already noted in https://github.com/zigpy/zigpy/issues/1640.
For now, please ignore the message.

closed 2025-10-06 by TheJulianJES