zigpy/zigpy #1661: '_TZ3000_nkkl7uzv' has incorrect direction

wire67, 2025-09-04:
Logger: zigpy.device
Source: runner.py:156
First occurred: 30 August 2025 at 19:46:17 (93 occurrences)
Last logged: 21:32:04

Cluster 0x000a on <Device model='TS0207' manuf='_TZ3000_nkkl7uzv' nwk=*** ieee=*** is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster). Please report this here: https://github.com/zigpy/zigpy/issues/1640

Hedda, 2025-09-08:
> Please report this here: [#1640](https://github.com/zigpy/zigpy/issues/1640)

zigpy/zha devs collect incorrect direction warning messages in a separate issue so you should post there instead and then close this:
 
- https://github.com/zigpy/zigpy/issues/1640

wire67, 2025-09-08:
https://github.com/zigpy/zigpy/issues/1640#issuecomment-3265536969