zigpy/zigpy #1662: Cluster 0x000a on <Device model='TS0001' manuf='_TZ3000_skueekg3' nwk=0x9F40 ieee=a4:c1:38:bb:97:03:a2:46 is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster)

f-wijnoltz, 2025-09-07:
Logger: zigpy.device
Source: runner.py:156
First occurred: 09:51:29 (27 occurrences)
Last logged: 10:25:16

Cluster 0x000a on <Device model='TS0001' manuf='_TZ3000_skueekg3' nwk=0x9F40 ieee=a4:c1:38:bb:97:03:a2:46 is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster). Please report this here: https://github.com/zigpy/zigpy/issues/1640
Cluster 0x000a on <Device model='TS0001' manuf='_TZ3000_skueekg3' nwk=0x4FA6 ieee=a4:c1:38:08:8b:ec:59:d8 is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster). Please report this here: https://github.com/zigpy/zigpy/issues/1640
Cluster 0x000a on <Plug_v3_NoGP model='TS011F' manuf='_TZ3000_ww6drja5' nwk=0x0BDA ieee=a4:c1:38:fc:4f:0e:bd:dd is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster). Please report this here: https://github.com/zigpy/zigpy/issues/1640
Cluster 0x000a on <Plug_v3_NoGP model='TS011F' manuf='_TZ3000_ww6drja5' nwk=0x2A78 ieee=a4:c1:38:a2:56:f9:a6:f4 is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster). Please report this here: https://github.com/zigpy/zigpy/issues/1640
Cluster 0x000a on <Device model='TS0001' manuf='_TZ3000_skueekg3' nwk=0x8AFB ieee=a4:c1:38:02:61:8d:c5:92 is_initialized=True> has incorrect direction (got <Direction.Client_to_Server: 0> for <ClusterType.Server: 0> cluster). Please report this here: https://github.com/zigpy/zigpy/issues/1640

wildpump, 2025-09-16:
Ok, Danke werde ich machen. Leider verfüge ich nicht die Berechtigung zum Schliessen diesen Beitrags.

TheJulianJES, 2025-10-06:
Thanks for the report! Devices are already noted in https://github.com/zigpy/zigpy/issues/1640.
For now, please ignore the message.